jgyates/genmon #517: GenPI loosing and resetablishing connection to Generac Controller since software update 16.20

wb9rro, 2021-03-02:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

wb9rro, 2021-03-02:
Time: 09:12:11 AM
Date: Tue 02-Mar-2021
Generator Monitor communications with the controller has been restored at Vines Senior Homes

Monitor : 

    Generator Monitor Stats : 
        Monitor Health : OK
        Controller : H-100 Controller, PM-DCP Release 4.2A1, HW00
        Run time : Generator Monitor running for 0:49:32.
        Power log file size : 0.01 MB of 15.00 MB
        Generator Monitor Version : V1.16.00
        Update Available : No

    Communication Stats : 
        Packet Count : M: 1285, S: 824
        CRC Errors : 0 
        CRC Percent Errors : 0.00%
        Timeout Errors : 460
        Timeout Percent Errors : 35.80%
        Modbus Exceptions : 0
        Validation Errors : 0
        Sync Errors : 0
        Invalid Data : 0
        Discarded Bytes : 0
        Comm Restarts : 0
        Packets Per Second : 0.71
        Average Transaction Time : 0.0360 sec

    Platform Stats : 
        CPU Temperature : 122.00 F
        Pi Model : Raspberry Pi 4 Model B Rev 1.1
        Pi CPU Frequency Throttling : OK
        Pi ARM Frequency Cap : OK
        Pi Undervoltage : OK
        CPU Utilization : 1.14%
        OS Name : Raspbian GNU/Linux
        OS Version : 10 (buster)
        System Uptime : 0:44:39
        Network Interface Used : wlan0
        WLAN Signal Level : 27 dBm
        WLAN Signal Quality : 37/100
        WLAN Signal Noise : 0 dBm
        WLAN ESSID : Vines2
        System Time : Tuesday March 2, 2021 09:12:10

Time: 09:09:50 AM
Date: Tue 02-Mar-2021
Generator Monitor is not communicating with the controller at Vines Senior Homes

Monitor : 

    Generator Monitor Stats : 
        Monitor Health :  Not receiving data. 
        Controller : H-100 Controller, PM-DCP Release 4.2A1, HW00
        Run time : Generator Monitor running for 0:47:11.
        Power log file size : 0.01 MB of 15.00 MB
        Generator Monitor Version : V1.16.00
        Update Available : No

    Communication Stats : 
        Packet Count : M: 1213, S: 775
        CRC Errors : 0 
        CRC Percent Errors : 0.00%
        Timeout Errors : 437
        Timeout Percent Errors : 36.03%
        Modbus Exceptions : 0
        Validation Errors : 0
        Sync Errors : 0

wb9rro, 2021-03-02:
This problem has started since upgrading software to 16.20. Re-ran software update to 16.00 (was this a rollback??). Still having issue. Used the update software on the status page to do updates.

jgyates, 2021-03-02:
The version number issue is described here:  https://github.com/jgyates/genmon/issues/515

Regarding the timeout you are seeing, I would need the following:

1) are you using TCP or direct serial
2) I will need to see your log files (submit them on the about page)

Beginning with V1.16.0 there is one change that could be relevant however the percentage of errors is a vote in the opposite direction. Your logs would shed more light on this.

vonstorch, 2021-03-02:
I started experiencing messages that are System Health: Not receiving data as well. Started this morning after upgrading to 16.20 and continued after the update to 1.166.01. I'm using a direct serial communication. I will send my log files as well. Thanks for you review.
Bill

jgyates, 2021-03-02:
Looks like the modbus change I made is the culprit. I am working on a fix now. The change works on powerzone, but not h100. I am working on a fix that will hopefully work on both. I will keep you posted.

wb9rro, 2021-03-02:
Jgyates,

Thank you for your response. I am directly connected via serial cable from the genpi to the H-100 controller (front panel serial connector). Can you point me to how to collect the log files you wish to see. I have filezilla.  I have been seeing some issues with the wifi conection to my network prior to this release. I will often see the genpi on the network but it won’t allow my connection. But it does connect to send me emails when the generator runs. 

 

James A Compton

Chief Operating Officer

Vines Senior Homes, LLC

847-922-9800 (cell)

 <mailto:jim@vines-seniorhomes.com> jim@vines-seniorhomes.com

wb9rro, 2021-03-02:
Thanks,

I’ll standby. Let me know if you need something from me.

 

James A Compton

Chief Operating Officer

Vines Senior Homes, LLC

847-922-9800 (cell)

 <mailto:jim@vines-seniorhomes.com> jim@vines-seniorhomes.com

jgyates, 2021-03-03:
I have checked in V1.16.02 that should correct the problem. Let me know if you have any issues. 

You can submit your logs and registers on the About page if your outbound email is setup with genmon. Once you update to the latest it would be helpful to get this data.

jgyates, 2021-03-03:
I am going to close this issue for not but feel free to post to this thread if you have any additional questions or data.

closed 2021-03-03 by jgyates

vonstorch, 2021-03-03:
I'm still seeing the issue after the upgrade to V1.16.02. I've submitted the registers and logs.

liltux, 2021-03-03:
@vonstorch I am not seeing any communication issues with the H100 over TCP.  That maybe the difference, but have you updated your raspberry pi recently through apt-get?  may try that.  just a suggestion.  I doubt you need a reboot, but may help as well.

jgyates, 2021-03-03:
@vonstorch, This thread is related to an issue specific to the H100 industrial controller. You submitted logs for an Evolution 2.0. I suggest your create a new issue for your problem. I have an Evolution controller and see no issues. My first guess would be to check your cabling.

skipfire, 2021-03-03:
I did have an instance of the same on Evolution 2, if a thread is posted for it I will follow that as well.  Though it was only on the 1.16.20 that I saw it.

wb9rro, 2021-03-03:
Upgraded and set you logs. Web interface dropped out when gen started. Putty was still active.  Web interface came back online after several min.Jim wb9rroSent from my Galaxy Tab A

Closed #517.

jgyates, 2021-03-04:
Got your logs. looks like things are working now. Let me know if you have any other issues.

vonstorch, 2021-03-04:
System working great after the update and OS apt update, plus reboot, No longer receiving "System Health: Not receiving data". Not sure why this effected the Evolution 2 Controller.

wb9rro, 2021-03-04:
So far so good. Thank you so much for your support!

 

James A Compton

Chief Operating Officer

Vines Senior Homes, LLC

847-922-9800 (cell)

 <mailto:jim@vines-seniorhomes.com> jim@vines-seniorhomes.com