jgyates/genmon #518: Having issues getting new version to load

bdwilson, 2021-03-04:
Things have been working fine for over a year. I upgraded code to 1.16.x recently, but also had a power outage yesterday where the generator was on for an hour.  When I load the genmon page, the dials/graphs do not adjust as they should, monitor page never loads, I see errors on the bottom bar. Attempts to refresh the page cause it not to load at all - I get the "loading generator monitor" page and it never loads. I've rebooted several times, saw there were issues on 1.16.x versions with things not working (closed issues), so I tried going back to 1.15.x.  That version didn't work either - same issues, but now resulted in error in genmon.log:

2021-03-04 08:49:15,574 : Error in GetAveragePower: time sequence error

So I'm now back 1.16.03.  I've also tried physically power cycling the unit. 

I am powering my unit off of the generator vs. battery, but again, this hasn't changed - I get undervoltage errors, but things have always worked fine.  I do have a cloned SD card in the unit from my Sunday backup which will be my last resort to try, but requires me removing & opening it up (which I'd prefer to not do but will if we need to debug further).

What got me down this path was that MQTT wasn't working properly (my automation didn't know power was off yesterday), otherwise, I wouldn't have been looking.
[genmon_logs.tar.gz](https://github.com/jgyates/genmon/files/6084174/genmon_logs.tar.gz)

jgyates, 2021-03-04:
You may have a corrupted power log (/etc/genmon/kwlog.txt). Backup all of your files on the About page, then either delete the file /etc/genmon/kwlog.txt or you can click the "Reset Power Log and Fuel Estimate" on the Maintenance page. Both actions will reset the power log. This will reset your fuel estimate if you are using that feature. If you make a note of how much fuel is in your tank you can compensate for this by adding the amount of fuel left in your tank to the "Subtract Fuel" setting on the advanced page. This settings tells genmon to subtract the given number of gallons from an assumed full tank, then subtract the amount used in the power log. 

Once you reset your power long, then restart genmon. If this does not fix the problem let me know.

The error "Error in GetAveragePower: time sequence error" is an error from previous versions. That error check was removed a sometime last year as additional checks were added to try to compensate for corrupted power logs.

If you want to attach your power log here (from your backup) I can look at it and try to duplicate the issue.

bdwilson, 2021-03-04:
Deleting both logs via command line and restarting seemed to fix it. Thanks!  

Where are the advanced settings?

jgyates, 2021-03-04:
Double click the gear icon in the top right of the web interface will show the advanced settings.

closed 2021-03-04 by jgyates