jgyates/genmon #519: SMTP stopped working

msyoung2001, 2021-03-08:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Prior to one of the last updates my email alerts were working.  For the last few weeks I have been getting the weekly test run notices and when I try to test the SMTP I get an error message,

### Actual Behavior

No emails are coming and an error is displayed when testing the SMTP setup.  An Error occurred: Error Initializing SMTP library: [Errno -2] Name or service not known.

### Steps to Reproduce (including precondition)

Going into settings and clicking on the Test Email Settings button.

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: 17kw air cooled
- Genmon Logs: Submit via the About page - Done
- Genmon Version: 1.16.03

THANKS!

jgyates, 2021-03-08:
Nothing has changed recently in the mail support. Submitting your logs works only if outgoing email is working. I will need this info to try to diagnose what is happening:

what are your settings (i do not need your exact email address or password, but I would need your SMTP mail server your are using and port number)

if possibly the contents of your /var/log/mymail.log file.

It is possible that you were using an older address like smpt.google.com which is now smtp.gmail.com. It is possible that your mail servers address has changed, but that is only a guess based on the info above.

jgyates, 2021-03-08:
The mail support in /genmon/genmonlib/mymail.py has not been updated for 3 months and that was only a minor change at that.

msyoung2001, 2021-03-08:
I use 365 and am using smtp.office.com with port 587 and SSL on and Disable TLS off.  I have verified that the SMTP server is right as well as email and password.  My main question is what exactly that message means?  Does it mean it can not resolve the FQDN or that it can not contact the server?

My only other thought is if MS changed something for SMTP relay but I have other devices working the same as this is so not sure on that..

zekyl314, 2021-03-08:
I would check your settings again, smtp.office.com doesn't appear to exist (Maybe it did in the past).  I see that smtp.office365.com does exist however.

jgyates, 2021-03-08:
try using smtp.office365.com instead

https://support.microsoft.com/en-us/office/pop-imap-and-stmp-settings-8361e398-8af4-4e97-b147-6c6c4ac95353

jgyates, 2021-03-08:
ditto on @zekyl314 comment

msyoung2001, 2021-03-08:
wow...  face palm..   yep that was it!  changed to smtp.office365.com and turned off ssl and away we go!  Now I need to go find out what and when this changed and why some of my other devices are still working!  Thanks for the help!!  Love this app!

jgyates, 2021-03-08:
No problem, I am glad it is working for you. Thanks for the assist @zekyl314

closed 2021-03-08 by jgyates