jgyates/genmon #520: Clear Alarms

raymondjclark, 2021-03-14:
I was wondering if the ability to clear alarms might be coming in a future request. I tried to remote start a generator this morning after a long cold snowy night. I got an oil pressure low alarm. I know it is because of the cold. Since I cannot remotely clear the alarm it means a trip to the remote location.

Thanks

jgyates, 2021-03-14:
this functionality is supported on controllers that appear to support this functionality (Evolution Liquid Cooled, some industrial units). in short if I knew how to implement this on Evolution air cooled units I would add this.

closed 2021-03-14 by jgyates

grsthegreat, 2021-03-17:
generacs new roll out on mobilelink supposedly can clear fault codes in air cooled units. or so i have been told.

liltux, 2021-03-17:
@grsthegreat maybe, but this project is reverse engineered.  So we would need to be sniffing the modbus or IP packets at the same time Mobile link clears the fault to get the modbus information required to do this.  I doubt Generac will tell us directly how this is done.

That being said, if anyone was willing to allow some sniffing to be done on their air cooled unit when the fault was being cleared by mobile link, that would be helpful.