jgyates/genmon #521: Reported frequency is 10x the actual value on Evolution 1.0 (hw v1.06, fw v1.15)

joecotton, 2021-03-17:
### Expected Behavior

Frequency should be approximately 60 Hz.

### Actual Behavior

Frequency is reported between 595.00 and 605.00 Hz

### Steps to Reproduce (including precondition)

When generator is running.

### Screenshot or Pictures relating to the problem (if possible)

{Note frequency report}
[image]

### Your Environment

- Generator Model: Generac Synergy 20kW, model G0070400, Controller Hardware v1.06, Controller Firmware v1.15
- Genmon Logs: Submitted Registers on 2021-Mar-17
- Genmon Version: V1.16.03

joecotton, 2021-03-17:
Possibly related to #473

jgyates, 2021-03-17:
This may be Synergy specific. I don't have a lot of register examples for the Synergy model. This is any easy fix. I will let you know once a fix is checked into to test with.

joecotton, 2021-03-17:
If you need any specific testing or diagnostic data dumps for Synergy, let me know.

jgyates, 2021-03-17:
I got your registers, that helped a lot. 
I checked in V1.16.04. If anyone has a Synergy model and this causes the frequency to be displayed incorrectly after this change let me know.  It could be firmware version specific but until I have a case for that I am going to assume this is the same for all Synergy models.

Thanks for reporting this.

joecotton, 2021-03-17:
I tested this and on my setup the issue is cleared.

closed 2021-03-17 by jgyates