jgyates/genmon #523: [Feature Request] Ability to control update alerts through Pushover

danodemano, 2021-03-19:
I have a loud and obnoxious alert setup via the Pushover add-on so I can keep tabs on the generator status. I also have update alerts enabled so that I know when to update genmon. I had assumed these only would come via email but they also come via Pushover which plays that obnoxious sound. If possible I would like there to be an option to only send these via email instead of the additional channels. Thanks!

[image]

jgyates, 2021-03-19:
I will look into this however one solution is to only use pushover for software update noticides and remove the "info" category from email notifications on the Notifications page in the web interface. Removing the info category will stop the following email notices: software update, software starting email, generator changing states if not in alarm or warning (i.e. exercise, stopping, etc), daylight savings time change.

This change will require modification of the underlying notify code in the project which will effect a few of the add on modules so I just need to weight the benefit vs the maintenance cost.

danodemano, 2021-03-19:
Thanks. No big deal if it doesn't work out. I know I can do it that way but I get so many emails I wanted to use pushover to really get my attention.

jgyates, 2021-04-06:
I will add this to the list.

https://github.com/jgyates/genmon/blob/master/feature_requests.md

I will update this thread once it makes it into the code

closed 2021-04-06 by jgyates

jgyates, 2021-11-24:
This feature has been added to V1.18.01 see the release notes for more details. Let me know if you have any issues or questions,

https://github.com/jgyates/genmon/blob/master/changelog.md