jgyates/genmon #524: SerialTest and SocketTest Fail

cmgsupport, 2021-03-21:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Expected a pass when the port is wired in loopback.  I am using a WiFi serial server.

### Actual Behavior

Received a fail.  Maybe this is a python version issue, but can be corrected with either of the following:

`ReceivedString = ReceivedString.decode("UTF-8")`
or
`if TestString != ReceivedString.decode("UTF-8")`

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2021-03-21:
I am unable to duplicate this issue on a raspberry pi with both python 2 and 3. Can you provide more details regarding your operating system and python version? What does "python -V" show. Have you installed the python serial libraries in the install process? You mention that you receive a fail but you do not show what failure. Can you provide more info on the failure?

cmgsupport, 2021-03-21:
Ubuntu 20.0.4.2 LTS (GNU/Linux 5.4.0-1029-raspi aarch64)

The failure was "FAILED: Sent data does not match receive. Received 14 bytes"

python -V says 2.7.18 but I ran the command with python3 sockettest.py -a 10.0.100.56 -p 8899

I aslo did this with serialtest.py and using socat to make a virtualcom port.

Pyserial 3.5 is installed.

I have larger issues as well, can't get the startgenmon.py going.   Issues parsing config.  Start a different thread?

jgyates, 2021-03-21:
This thread is fine as this appears to be related to Ubuntu defaults vs raspbian

I connected a loopback and can see the issue now.  It is a python3 issue. I checked in a fix.

Regarding the startgemon.sh script. I assume you intended to type startgenmon.sh. 

I will test this on ubuntu and see if I can see an issue. Do you have a error or line number when executing the script?

cmgsupport, 2021-03-21:
Great, thanks for your help.  Yes, typo above.

First, I'm just in a test environment right now, generator is at another location.  This should still fire up the server correct?
Here's the issue:

ubuntu@ubuntu:~/genmon$ sudo ./startgenmon.sh -p 3 start
Using Python 3.x...
Starting genmon python scripts
Error validating config. Retrying..
Error validating config file, Exiting

sample contents of /var/log/genloader.log

ubuntu@ubuntu:~/genmon$ vi /var/log/genloader.log
2021-03-21 10:27:36,084 : Error in MyConfig:ReadValue: module: 'list' object has no attribute 'find' : myconfig.py:85
2021-03-21 10:27:36,086 : Error in MyConfig:ReadValue: enable: 'list' object has no attribute 'find' : myconfig.py:87
2021-03-21 10:27:36,086 : Error in MyConfig:ReadValue: hardstop: 'list' object has no attribute 'find' : myconfig.py:87
2021-03-21 10:27:36,087 : Error in MyConfig:ReadValue: conffile: 'list' object has no attribute 'find' : myconfig.py:85
2021-03-21 10:27:36,087 : Error in MyConfig:ReadValue: args: 'list' object has no attribute 'find' : myconfig.py:85
2021-03-21 10:27:36,088 : Error in MyConfig:ReadValue: priority: 'list' object has no attribute 'find' : myconfig.py:91

jgyates, 2021-03-21:
These errors indicate an issue with your /etc/genmon/genloader.conf file. The program is trying to validate your config files, then it attempt to copies them again and it still fails. Are your running the program from the repository directory? If the full repository is not present it will not work. Do you have a file /etc/genmon/genloader.conf and does it look like the file in the github repository /conf/genloader.conf?

jgyates, 2021-03-21:
I can duplicate this with python3 on ubuntu, but not python2. I will see if I can come up with a fix.

cmgsupport, 2021-03-21:
> This thread is fine as this appears to be related to Ubuntu defaults vs raspbian
>
> I connected a loopback and can see the issue now. It is a python3 issue. I checked in a fix.
>
> Regarding the [...]

You only need to add one of those lines, I see you put in both.  Maybe the if statement is the only one to add.

cmgsupport, 2021-03-21:
Will the fix for the serial test be needed in any of the other packages?

jgyates, 2021-03-21:
I don't know yet. I have tested this with python3 in the past however it appears that things have changed. Raspbian still defaults to python 2.x so that is what gets the most testing.

cmgsupport, 2021-03-21:
Sounds like I should just stick with what works. :)

jgyates, 2021-03-21:
python 2.x has 99.99% of all of the testing and use cases. 3.x has very little. I do resolve 3.x issues when I am aware of them. I used to run 3.x in my production but there is an issue with python threads starting properly in a boot cron job that make that unusable for me so I stopped doing that.

jgyates, 2021-03-21:
I found the issue. The config file /genmon/conf/genloader.conf had a duplicate section. python 3.x did not allow that, but python 2 was OK with this. Delete the file /etc/genmon/genloader.conf the update the software via this command:

     ./genmonmaint.sh -u

This will checkout the latest in github. Once you start it up again the file will be copied from /genmon/conf to /etc/genmon

Let me know if you have any other issues.

cmgsupport, 2021-03-21:
Great, thanks!  Will give it a whirl.

cmgsupport, 2021-03-21:
I'm running on Raspbian and Python2 as well now.  I had one issue regarding SMTP setup.  I was able to get around this by editing the mymailconf.py file directly, but the UI doesn't like newer unusual domains.  I needed to use a .support address and the validation wouldn't let me.

jgyates, 2021-03-21:
The domain and other parameter checking in the web interface just highlights the UI red if there is a suspected problem. You can still save any domain name. Which parameter specifically did you try to use a .support domain?

cmgsupport, 2021-03-21:
oh, haha, well look at that.  I just assumed it wouldn't take since it was red.  Never mind! :)

jgyates, 2021-03-21:
I am closing this issue for now but feel free to post here on any issues related to this thread.

closed 2021-03-21 by jgyates

cmgsupport, 2021-03-21:
I updated the genloader.conf and I get this result when starting

```
ubuntu@ubuntu:~/genmon$ python3 ./startgenmon.sh -p 3 start
  File "./startgenmon.sh", line 7
    genmondir="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
                      ^
SyntaxError: invalid syntax
```

jgyates, 2021-03-21:
startgenmon.sh is a bash script, not a python program.

      bash ./stargenmon.sh -p 3 restart

If you delete the file /etc/genmon/genloader.conf then run this:

      ./genmonmaint.sh -u

you should be set. No need to copy anything if you can update it directly using git.

jgyates, 2021-03-21:
The commands in order are:

     rm /etc/genmon/genloader.conf

     ./genmonmaint.sh -u

      ./stargenmon.sh -p 3 restart

You don't need the 'bash' before if the files have executable permission. The initial install script should have set the permission. If you installed this via git and installed the libraries via genmonmaint.sh then it should work. Others have used this with ubuntu with no issues.

cmgsupport, 2021-03-21:
ah yes, duh, python in front of ./startgenmon.sh - oops