jgyates/genmon #525: Huge timeout errors

jcherney, 2021-03-21:
Not sure if this is a software issue or hardware. Received 2 notifications this week that the Generator Monitor comms were restored. After looking at the logs, the timeout errors which are usually nonexistent or 1, are 452.

Any thoughts why this might be taking place recently?

just sent logs and registers.

jgyates, 2021-03-21:
bad cable, restart the pi or bad voltages on the pi, but these are just guesses. Nothing on the serial comms stuff has changed recently.

jcherney, 2021-03-21:
Thanks. I’ll try pulling the power first and see what happens.

alabamatoy, 2021-03-21:
Same thing here.  Brand new install, almost 100% timeout errors.

djrallard, 2021-03-21:
Check your ground connection. Two years ago, when I first installed my GenMon, I wasn’t properly using the ground from the generac controller and I was getting a lot of timeout errors. Once I resolved the grounding issue, I haven’t had another timeout.

jgyates, 2021-03-21:
I agree with @djrallard, a bad ground connection will make your throughput at best marginal and at worst it will cause damage to the transceivers.

jgyates, 2021-03-27:
I am going to close this issue for now. If you have any additional data or questions feel free to post to this thread.

closed 2021-03-27 by jgyates

jcherney, 2021-03-28:
Sorry, forgot to respond. I pulled the power for 15-20 seconds and it cleared up my issue.