jgyates/genmon #526: genmon behind apache2 proxy

alabamatoy, 2021-04-04:
I am trying to run genmon behind an Apache2 proxy.  I probably have the proxy configured incorrectly because Apache2 is very complex and the config files are likewise, but...

The proxy works, but all I get displayed in the browser is the "loading generator monitor" and the swirling circle. The actual details never appear, so it appears to be a javascript issue.  If I go directly to the genmon node, it works correctly.  View source produces the same HTML content whether via the proxy or direct.  I dont see any errors in the Apache logs.

Any suggestions?

Apologies if this is not the correct way to ask about this...

jgyates, 2021-04-04:
The javascript uses the URL of the server (obtained from the URL used in the browser) to make AJAX calls. If the proxy and genserv.py are not on the same IP address, then this will not work as the AJAX calls made by the javascript will fail unless the proxy also is a proxy for AJAX calls.

alabamatoy, 2021-04-04:
Wow, OK, reading about proxying Ajax calls, apparently theres not really a documented solution, only some hacks that kinda work.  Sometimes.  Apparently proxying the AJAX calls through Apache is quite difficult.  So I do not know how to configure "the proxy also is a proxy for AJAX calls."

If I could understand what needs to be aliased so the javascript would work, I can use Apache aliasing to correct it, I think.  As in:
`<VirtualHost *:443>
    ServerName example.com
    Redirect 301 / http://www.example.com/
</VirtualHost>`

Running genmon behind a proxy seems to be the *right* way to deploy it.  Any help would be appreciated...

jgyates, 2021-04-04:
Is apache currently running on the same computer as genmon? If it is not then I would try that first.

This link may have some more info that is helpful:

https://www.codementor.io/@abhishake/minimal-apache-configuration-for-deploying-a-flask-app-ubuntu-18-04-phu50a7ft

The info in the link above should work as long as apache is running on the same computer as genmon.

If your intent is to expose the web interface to the internet instead of a private LAN, my recommendation would either:

1) Use a VPN. It is not recommended to expose the web interface to the raw internet as the internal Flask server is not robust or built with security in mind. This is how I use it remotely.
2) Use apache with Flask, see the link above for more details.

The current implementation uses the Flask internal web server, which is not recommended for production oriented sites. A proxy would still use the internal server, it would just be redirected via proxy.

Both of these solutions would still put everything running on the pi so you should not have to worry about any proxy issues.

alabamatoy, 2021-04-04:
"Is apache currently running on the same computer as genmon? If it is not then I would try that first."

Not possible without lots of rework.  Generator is remote from Apache with about 200' Cat6 from router to Pi inside the genset.

The Pi is not exposed to internet, and I dont intend to do so.  The Apache is exposed to internet, its pretty heavy iron, gets hammered by script kiddies all the time.  The proxy currrently demands auth to get to the Pi/genmon, and that works.  Its all SSLed by the proxy.

The pain is that everything works, right now....except for the AJAX issue...So close....

jgyates, 2021-04-04:
you could modify lines 18 and / or 19 in genmon.js to use the IP address of the pi

https://github.com/jgyates/genmon/blob/master/static/genmon.js#L18

then you would need to run the closeure_compiler.py program in the /genmon/static/ folder to compress the javascript. or use the /index_verbose.html page (this pages is uncompressed, or without the removal of extra whitespace, javascript)

This is sort of a hack, but it is an option.

Of course you could just expose a single port on the pi instead of the entire IP.

alabamatoy, 2021-04-05:
"you could modify lines 18 and / or 19 in genmon.js to use the IP address of the pi" the following works, using index_verbose.html, going directly to the pi, but not through the proxy:
```var pathname = "http://192.168.1.130:8000/";```
```var baseurl = pathname.concat("cmd/");```

Logically I can see why it wouldnt, because the javascript is executed on the client, and the client cannot get to 192.168.1.130, right?

"Of course you could just expose a single port on the pi instead of the entire IP."  That's what I am doing.  The proxy is pointed at the internal address 192.168.1.130:8000 so its only exposing the one port....

jgyates, 2021-04-06:
yes, that will not work if the browser and genmon are not on the same network.

you can load genserv.py on the same system as the apache server. you would need to add "-a 192.168.1.130" on the command line to tell it that genmon.py is on that IP address:

      sudo python genserv.py -a 192.168.1.130

You can add this to the crontab on the apache box to start it on boot. the same config file on the pi at /etc/genmon/genmon.conf would also need to be on the apache server (/etc/genmon/genmon.conf). This would make genserver (which accepts the ajax requests) reside on the same server as apache. Then genserver.py would connect to genmon at 192.168.1.130. put the entire genmon repository on the apache box. You would need the python libraries installed on the apache server 

     ./genmonmaint.sh -i

genserv.py by default uses not secure http and port 8000. This is configurable via the web interface / config file. I would recommend getting genmon/genserv setup with the ports you want then copy the /etc/genmon/genmon.conf file over to the apache box.

I hope this helps.
Answer "no" to everything except installing the libraries.

alabamatoy, 2021-04-06:
OK, I dont like to give up... I kept beating on this, and got it working.  The two lines in genmon.js were changed as follows:

    var pathname = "https://myDNSname.com:8080/";
    var baseurl = pathname.concat("cmd/");

I renamed index.html to something else, and renamed index_verbose.html to index.html and it began working.  No other changes, seems to be all working correctly.

The Apache config is as follows:

    <VirtualHost _default_:8080> 
        SSLEngine on
        SSLCertificateKeyFile /etc/letsencrypt/live/myDNSname.com/privkey.pem
        SSLCertificateFile /etc/letsencrypt/live/myDNSname.com/fullchain.pem
        #ProxyRequests Off
        ProxyPass "/" "http://192.168.1.130:8000/" timeout=10
        ProxyPassReverse "/" "http://192.168.1.130:8000/" timeout=10
        <Proxy *>
        		Order deny,allow
        		#Allow from all
        		Authtype Basic
        		Authname "Password Required"
        		AuthUserFile /etc/apache2/.htpasswd
        		Require valid-user
        </Proxy>  
    </VirtualHost> 

"8080" in the above config can be any of a number of suitable ports.  the .htpasswd file enforces basic long/pw behind SSL encryption.  There's probably a way to get it onto 443 through a virtual directory (as in https://myDNSname.com/mygenerator ) but I probably wont chase that for now.  Apache configuration is like a mystical black art and I am just entering Hogwarts.

I respectfully suggest that you add a simple config item somewhere in genmon config for "will genmon be behind a proxy" and if yes, enter the external name:port of the server, and then pass that to the pathname variable.  But you probably need to throw up a big warning about not allowing the world to beat on the genmon, ie authentication is essential.

jgyates, 2021-04-06:
I added this (2nd) paragraph to the main wiki page:

https://github.com/jgyates/genmon/wiki#connectivity

Let me think on the proxy. I am not sure how to pass this info to the javascript other than manually writing the file, it is sort of a chicken and egg problem.

I do have adding apache support on the list of features. 

Also, if your modified html files are on the pi then a software update will overwrite them. Software updates only occur when the user selects this on the About page.

alabamatoy, 2021-04-06:
> Also, if your modified html files are on the pi then a software update will overwrite them. Software updates only occur when the user selects this on the About page.

Yes, that's why I havent updated it for a while since I have been tinkering with the code.

> I am not sure how to pass this info to the javascript

Cant you use a env file for that?  I think it requires inclusion of a dotenv package.  I know only enough about Javascript to get us both into trouble.....

BTW, close this at your discretion...

jgyates, 2021-04-06:
" I know only enough about Javascript to get us both into trouble....." that makes two of us

I will look into that, thanks for the tip.

closed 2021-04-06 by jgyates

alabamatoy, 2021-07-10:
As noted above in early April, you said you were going to look into a config option for "run behind proxy" or similar and then require the user to configure the FQDN and port for the proxy.  I know you have a lot of 2dos, but did you ever look into doing this?

Its a PITA to update as it stands (See my post above from May 11).  If the option for proxy could be added (with default = no proxy) and a system var to replace the static value for pathname if proxy is selected, this would allow updates to be applied without so much fuddling about with the files in order to update.

The Apache2 config is also listed above, which could be a good addition to the wiki.  I will assist by writing up a wiki entry for the apache2 config if that would help.  

Please consider the proxy alternative for implementation.

jgyates, 2021-07-20:
This is a non-trivial change. It requires modifying the methods used to load the main web pages (change from a static load to a form load in the flask api).  This would be required to load the page while defining a variable for the page. Since the change will impact the loading of the web interface I am hesitant to make this change as it will potential effect everyone who uses this project. If there is an error in the code (or some unexpected behavior with the flask library) and the web interface fails to load this would require everyone to re-install as the mechanism to update the software would be broken. Considering the above and this feature only appears to impact you this is not something I am considering soon if at all. Sorry to disappoint but the incentive for the group is just not there at this point.