jgyates/genmon #527: What determines Fuel Level Low event?

liltux, 2021-04-07:
@jgyates what triggers a Fuel Level Low Event?  We had some network issues last week and this week I found our RPi was locked up.  I restarted the RPi and everything was as it should be and functional.  Shortly there after there was an alert/message sent out about low fuel level.  It is at 79%.  Is this hard coded somewhere?  I think this may be a value to be set per installation.  Your thoughts?

jgyates, 2021-04-07:
The fuel level low event is triggered by a fuel level reading of 20%  or less and another notice at 10% or less. Was the message and email or some other short message? Logs may provide more info on what is happening and why you got an incorrect notice.

The software currently determinants the fuel level by one of three ways depending on the installation.

1) Fuel sensor built in to controller (industrial or Evo Diesel)
2) External fuel sensor add on (tank utility or DIY tanks sensor)
3) Calculated fuel consumption (Evolution or Industrial)

I would look on the Monitor page an see if there are communication errors. In some situation communication errors can cause invalid register readings which could make case number 1 above read bad info.

jgyates, 2021-04-07:
also, if the notice is recieved from a message other than email then the state can be checked on the Maintenance page:

Maintenance - > Fuel Level State :  (OK or Low)

If this value is Low then a message is sent out via short message (non email)

liltux, 2021-04-07:
Ok,  I was thinking the same thing.  On the Industrial units, they should annunciate on their own for Fuel level warnings configured in the controller.  You could disable this alert for Industrial units, just a suggestion.  I did send my logs to you, maybe you will see something?  I did not see anything other than the event from the previous week.  Curious, could the mail message been stuck in limbo in SMTP and once rebooted and good network connection the message was able to send?  I have seen if there is a network problem, once corrected, you can get a flood of messages that were not delivered during the networking issue.

liltux, 2021-04-07:
And I got another message.  Hmmm.  It appears to be triggering 11 min. after a software reboot.  Almost to the T with the time stamp.  But no other issues I can see in the logs, or on the Generator.

jgyates, 2021-04-07:
What is the exact message state?

liltux, 2021-04-07:
[image]
[image]

liltux, 2021-04-07:
See above, approximately 11 min after restart of software I recieve a Fuel Level is Low.  I am only setup for Texts.

jgyates, 2021-04-07:
This is a bug. I can duplicate. I get a SMS low fuel but no email. I will let you know once I have a fix.

liltux, 2021-04-07:
Ok, thanks.  I wasn't sure.  Thank you sir!

jgyates, 2021-04-07:
I checked in V1.16.05. This should fix the problem. Let me know if you have any issues.

closed 2021-04-07 by jgyates