jgyates/genmon #529: Random System Restarts

condraka, 2021-04-08:
System has been working great and no problem running, reporting or control.  System is working great......except....

Recently the unit is randomly restarting and the only reason I know it's doing it is the email I get that it's restarting (see attached screen shots and attachments).  The unit ran all winter with no issues and just in the past month it's restarting for some reason and sending me an email that it did.  Why is it restarting?  It's random and happens at different times.

[image]
[image]

Using a RPi 3B+ attached to it is a RPi UPS V3 backup power module to hold power up while Generator starts up and switches over to itself during power outage.  Also have heat sinks with fans to keep board cool (See picture).

This one has be stumped.  Do you need any log files or additional information.  Seems like an update recently started this behavior.  

Thanks for this great alternative to the Generac Cellular system.  I had the original wireless communicator and when it died and could not get it fixed, I was not going to spend $$$ and Monthly Fees (come on Generac why not wireless).  So your system is far and beyond better than the OEM's.

jgyates, 2021-04-08:
Nice set up with the UPS.

Have you tried running a disk check on the SD card? You may have some corruption?

https://www.raspberrypi.org/forums/viewtopic.php?t=179361

Another thing to check is the Platform Stats on the Monitor page. This displays some Raspberry pi metrics about under voltages that may show a power problem if one exist.

Possibly check the integrity of your USB-C cable going to your power supply / UPS. You could have an intermittent connection.

You could try another SD card as a stab in the dark. I don't know of anything in the software that would cause a reboot to occur.

jgyates, 2021-04-08:
If you want to send your log files on the about page I could take a look to see if anything appears abnormal.

condraka, 2021-04-08:
Good stuff to check.  I have a secondary card with a saved image (I backup all my pi’s after each update and save an image to my PC.)
Will image another card and check the current card.  I’ll check stats.  I’m running a Mackertop 5.25v 3A power adapter ( my 3D printers run pi’s and most power supplies give the lightning bolt icon.  This supply is tops and no bolt icons....  good power.  

Is there a way to detect a power loss easily?  I checked the logs on the Pi and nothing  in There look out of the ordinary.  Maybe the UPS is dropping and coming back on.

Strange because it just runs great and out of the blue appears to restart....

Thanks for the help. Will report back after additional tests/checks

Carl E. Ondraka

jgyates, 2021-04-14:
I am going to close this issue but feel free to post here if you have any updates or questions.

closed 2021-04-14 by jgyates

condraka, 2021-04-14:
Ok.  It has not happened lately and wondering it it’s a wiring issue.......

Found the 110vac line going to the RasPi was loosing connection and over heating due to a loose wire.  Have an electrician fixing this.  I tied it up and so far no restarts.

Carl E. Ondraka

sarahmva, 2021-04-14:
One thing to always check in such situations is the power supply. I had a lot of issues with power supplies recently where they either died or could not provide enough power to run devices. In my initial attempt, I tried connecting the GenMon directly to the battery using a 12V to 5V converter but that just caused restarts every 30 seconds.

condraka, 2021-04-26:
It was a power issue due to a failing fuse on the generator side 110v supply for the RPi power adapter.

Question.....  the MakerFocus RPi 4 battery UPS has an I/O UART pin set available.....  could this be integrated to communicate to GenMon to report power failures to the RPi like my issue to report issues with supply?

Thanks

Carl E. Ondraka

jgyates, 2021-04-26:
Software already exist to provide UPS notifications. You should be able to setup the UPS and Pi to send emails on power failure events with this software:

https://melgrubb.com/2014/09/05/raspberry-pi-home-server-part-15power-failures/

You would need a 2nd serial port as genmon uses one serial port to communicate with the generator controller. The 2nd serial port would be connected to the UPS to monitor power events.