jgyates/genmon #531: Generator not turning off when clicking in "Stop Generator" and electrical utility is out

cuncil1, 2021-04-18:
Hello! First, thanks for Genmon, it is great! I have one question, in order to decrease the amount of propane used, I also connected the option to use inverter generators. The main problem I have now is that when electricity is out, the gas generator turns on and I cannot turn it off with the app. So, my question is, shouldn't the gas generator turn off when I turn it off with genmon even if the electrical utility is out?

### Expected Behavior

Generator should turn off when requested in genmon even if the electricity is out.

### Actual Behavior

Generator is not turning off after clicking on Stop Generator and OK to the message related to cooling down.

### Your Environment

- Generator Model: G0062600
- Genmon Logs: Submit via the About page (Submitted)
- Genmon Version: Version V1.16.05

jgyates, 2021-04-18:
There is a shutdown procedure in the operators manual that needs to be followed to shut down a running generator under load. This is the safe way to shutdown the generator while it is under load.

Stopping a running generator under load is not advised by generac as it could damage the generator. You need to remove the load first. Also, the stop command will work if the utility power is connected. The transfer switch will not switch back to the "line" state if power is not present on the line since this is how it is designed to work. In an actual outage you need to follow the steps in the manual to avoid damage to your generator and avoid bypassing safety measures the transfer switch provides.

I hope this helps. Let me know if you have other questions. I am going to close this issue but feel free to post to this thread on related questions.

closed 2021-04-18 by jgyates

liltux, 2021-04-18:
Well said @jgyates, and to add additionally.  When shutting down a generator under a load it also creates a brownout effect.  By doing so can cause significant damage to personal property in your home as well.  Damage appliances as well as sensitive electronics such as tvs and modems.

cuncil1, 2021-04-20:
Hello and thanks for the rapid response!

> Stopping a running generator under load is not advised by generac as it could damage the generator. You need to remove the load first. 

I installed the manual transfer switch for the small inverter generators between the Generac and the house. In this way, I change the power from the Generac to my small inverter generators first, taking the load off the Generac. After there is no load connected to the Generac, should the "Stop Generator" button stop the generator? I do not care if the Generac transfer switch switches back to utility or not, but I would like to turn off the generator without going into the roof after I take the load off.

Thank you again very much!

jgyates, 2021-04-21:
Unfortunately this is not something that the genmon software could do. All genmon does is issue commands to the controller. The controller has specific logic that will ignore commands that are not allowed.