jgyates/genmon #533: Outage notice delay

mefbgil, 2021-05-11:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Hello - new feature in V1.16.06, I can't see where in the UI or Conf files to set "outage_notice_delay" - can you point me to the right place?  Thanks.

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2021-05-11:
The UI setting is on the Advanced page. To view the Advanced page you must double click the gear icon in the upper right corner of the web interface. Then set the "Outage Notice Delay" parameter to the number of seconds to delay before a power outage counts as a reportable outage, then save your settings on the bottom of the page.

mefbgil, 2021-05-11:
Got it - thank you!  And thanks for adding this feature.

closed 2021-05-11 by mefbgil