jgyates/genmon #534: "Not Receiving Data"  followed by "OK" Message

ruggierir, 2021-05-18:
Not sure if this has been addressed before.  I have been using GenMon over a month now with the 3 temperature monitors from the RPi-pHat-Thermocouple project.  It has been working flawless up to now.  Over the last week or so, the GenMon has been sending via SMS text the ""Not Receiving Data" message, followed by the "OK" message 7 minutes later.  This goes on all day long.  I have updated to Version V1.16.06 to try and fix the issue, with no resolution.    I have just sent the logs and Registers from the "About" panel.  Other than those messages, everything else works great.

### Your Environment

- Generator Model: G0062410
- Genmon Logs: Submit via the About page
- Genmon Version: V1.16.06
- Firmware Version: 1.14
- Hardware Version: 1.05
- Boot Loader:  1.0
- EEPROM: 1.02

jgyates, 2021-05-18:
Based on your logs it looks like you are using serial, no serial over TCP. The messages you are recieving are the changing state of the genmon monitor. You can view this status real time on the Monitor page under Monitor - > Generator Monitor Stats -> Monitor Health. When this changes you get a message (other messages are generated from other sources). The "Not Receiving Data" message occurs when the received packet count does not change over an expected period of time. If you do end up receiving more data then the communication is deemed to be "restored" and the state of genmon changes to "OK", which generates another message. 

The behavior you describe indicates you have intermittent serial communications. I would guess that one of the following is wrong:

1) you have a marginal connection in your cabling.  Transmit, Receive and Ground must all be good stable connections.
2) You have some additional software on the pi trying to use the serial port. Have you installed any software recently?
3) you could have a bad SD card. Are you using a class 10 or better SD card and is it as new SD card? Bad sectors on the SD card can make strange things occur within the operating system.
4) your RS-232 transceivers on your controller are marginal (I doubt this but it is possible)
5) your pi is not grounded to the controller properly or you have power problems on the pi.

Do you see any errors on the Monitor page under Communication stats or platform stats? (You likely have timeouts occurring but do you have other errors?)

You log file /var/log/myserial.log has several errors that look like this:

2021-05-17 19:40:04,226 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120

This error occurs when the operating system tells the software that the serial port is not available any longer. If these errors are the cause of your issues (not the result) then this would lead me to thing that it was either 2, 3 or 5 listed above.

I hope this helps.

jgyates, 2021-05-18:
Also, there is a serial troubleshooting guide here: https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

ruggierir, 2021-05-18:
Thank You.  I will confirm each of these and reply back with results.  Thanks

jgyates, 2021-06-02:
I am going to close this issue for now. If you additional data feel free to post it to this thread.

closed 2021-06-02 by jgyates