jgyates/genmon #535: Engine Displacement & Rated Max Power for New Model Not Showing

yguy75, 2021-05-19:
**Expected Behavior**
Engine Displacement: 816 cc
Rated Max Power :       16 kW

**Actual Behavior**
Engine Displacement: 420 cc
Rated Max Power       : 7.5 kW

**Steps to Reproduce** 
Checking the Maintenace page shows the incorrect values.

This could be a new model and solution could be similar to existing issue #494
In Issue #494 the author said something similar to this.
New model information needs  to be added to “generac_evolution.py” or elsewhere
"16KW", "60", "120/240", "1", "816 cc", "240"] # Evo G0071760

Once the changes have been made, how do I update the Genmon application with the modified programs.

**Also of interest.** 
The initial run of Genmon did not display a Model Number or Serial number.
This was resolved by following issue # 482.
The version number was removed from “genmon.conf” .
After restarting Genmon, the model number was updated correctly.

**Environment**
Generator Model: G0071760
Genmon Logs: have been Submit via the About page
Genmon Version: V1.16.06

**Screenshot or Pictures relating to the problem**

[image]

jgyates, 2021-05-19:
@yguy75,

Thanks for the feedback.

I did not receive your registers. At a minimum would need to know the value of register 0019 on the registers page (single click the gear icon in the top right of the web interface) to make this change.

yguy75, 2021-05-19:
I resubmitted Register and Logs.
Maybe email isn't working.

Model Identity (EvoAC)
(0019)

HEX: 0016   DEC: 22 \| HI:LO: 0:22

jgyates, 2021-05-19:
I still did not receive any logs. I guess your outbound genmon email is not working. I just checked in an update. I did not change the version number so just go to the about page and click the update button. Let me know if this works for you.

yguy75, 2021-05-19:
Yes... That worked... Displacement and Max Power are OK.
Great.... Thank you very much.

We do receive email notifications from Genmon.
I assumed the registers would be sent.

Is there anything I should do to check the outbound email to you.

jgyates, 2021-05-19:
Not sure what is happening with the email. You could just send it manually. To to that use this URL:

substitute your IP address and port number on your pi below:

http://yourPiIPAddress:8000/cmd/support_data_json

Then copy and paste the data into an email and send it to generatormonitorsoftware@gmail.com

That will at least give me a test point with your registers.

I am going to close this issue but feel free to post here if you have any issues or questions on this topic.

closed 2021-05-19 by jgyates

yguy75, 2021-05-19:
Thank you very much for the quick response and modification.
You have created a great application with well written routines with easy to follow instructions.