jgyates/genmon #536: Enhancement Idea: Shutdown Button on Web Interface

Mykal451, 2021-05-20:
Michael Peters
May 20 2021

### Enhancement  Idea

I'm setting my neighbor up with a Genmon installation (thank you for such a great application). My neighbors are not Linux savvy people, in this case being able to gracefully shutdown the Pi via the Web interface would be a valuable addition to the app. Could you add a simple web based button to initiate "shutdown -h now" or "sudo poweroff", possibly with a confirmation dialog, or countdown indicator & cancel button in case of accidental clicking (since re-booting the PI requires going out to the generator, we live in the North East; snow, ice and cold).

Thank you,
-mp

jgyates, 2021-05-20:
I will look into adding this to the user interface. For what it is worth the following commands exist now:

      IPADDRESS:8000:/cms/restart    - this URL will restart the genmon software 

      IPADDRESS:8000:/cms/shutdown    - this URL will shutdown the pi

No confirmations are given.

jgyates, 2021-05-20:
I updated the above comment to correct a typo

      IPADDRESS:8000:/cmd/restart    - this URL will restart the genmon software 

      IPADDRESS:8000:/cmd/shutdown    - this URL will shutdown the pi

jgyates, 2021-05-22:
I checked in V1.16.08. There are now buttons on the bottom of the Advanced page for restart, reboot and shutdown. You can reach the advanced page by double clicking the gear icon in the top right corner of the web interface. Let me know if you have any issues or questions. I am closing this thread but feel free to post here if you have anything further on this topic.

closed 2021-05-22 by jgyates

Mykal451, 2021-05-23:
Thank you very much, amazing turn around on implementing this feature !! The warning dialog is perfect.
Regards,
mp