jgyates/genmon #537: Transfer a microSD card from a Pi-3B to a Pi-2b and work with no modifications.

yguy75, 2021-05-24:
**Question:**
Is it possible to transfer the microSD card from a Raspberry Pi-3B running Genmon to a Raspberry Pi-2B and NOT have to make any configuration changes?

Also: Could the SD card be transferred to a Pi-4 and work correctly?

In particular would the Serial Ports work correctly and not require re-configuring.

The Pi 2B could then be used as a good backup device.
The Pi-4B may be a future upgrade.

### Your Environment

Generator Model: G0071760
Genmon Logs: have NOT been submitted
Genmon Version: V1.16.08

jgyates, 2021-05-24:
@yguy75 

I believe the answer is yes with a few caveats. It is my understanding that with newer  versions of Raspbian this is possible but not with older version. This thread has more details:

https://www.raspberrypi.org/forums/viewtopic.php?t=260310

I have not personally tried this.

yguy75, 2021-05-24:
I'll run a test and transfer from a Pi3 to a Pi2.
Will get back to you in a few days.
Thanks.

skipfire, 2021-05-24:
@yguy75 The biggest with Pi2 is that Pi2 does not have WiFi, so you will have to be wired.  I have successfully swapped Pi2/Pi3 SD cards in the past in wired scenarios.  I have tried Pi3 to Pi4 with partial success on Buster, but the WiFi config did not work on both so I had to log in with keyboard and screen on the second unit and update the WiFi config manually and then was able to go back and forth.

liltux, 2021-05-24:
Just as @skipfire and @jgyates have said it should be fine.  One thing I have noticed between the different Rasbian versions, need to check the /boot/config.txt file.  Make sure the comments entered by Genmon are not in Pi specific sections of the file that may be ignored depending on which board you are using.  I have seen this happen.

yguy75, 2021-05-24:
Thanks for the advice @skipfire and @liltux.
I'm using wired ethernet, so don't need WiFi.
I will get back once I've confirmed if it works.

yguy75, 2021-05-26:
Cloned an SD card from a Pi3 and inserted it into a Pi2 in a test environment that was NOT connected to a generator.
**All test were successful...**

**One Exception:  
There was no "serial1" port in /dev/. 
Only "serial0"
Is this correct for a Pi2. ???**

pi@raspberrypi:~ $ ls -l /dev/ser*
**lrwxrwxrwx 1 root root 7 May 25 17:33 /dev/serial0 -> ttyAMA0**

[image]

Next will be a test connected to a Generator. Sometime in the future.

**Test Results from the Pi2 not connected to a generator.**
Serial Test - PASSED
pi@raspberrypi:~/genmon/OtherApps $ python serialtest.py
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
**PASSED! Loopback successful**

**Checked SerialConfig – All OK.**
Checking : BT service disabled: OK
Checking : Serial console command line removed: OK
Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console service disabled: OK

jgyates, 2021-05-26:
This is good info. It *should* work if the loopback test passed (assuming you have a loopback attached). Of course you would have to change the genmon settings page to /dev/serial0

I can't remember the names of the devices, but I do recall that the naming changed at some point along the way. This could be right.

yguy75, 2021-05-26:
No changes were made to the settings page and it did have /dev/serial0.
Therefore I assume the Pi2 is 100% compatible with the Pi3 for running Genmon.
We will test this Pi2 on a generator in the future.
Thanks for your help.
This has been a fun project for me.

closed 2021-05-27 by yguy75