jgyates/genmon #539: Modbus Exception and Error Receiving Slave Packet repeats every 8 minutes.

yguy75, 2021-05-27:
Actual Behavior and Observations:

1.	Mymodbus.log has a “Modbus Exception” and “Error Receiving Slave Packet” every 8 minutes
        BUT was 14 minutes when the generator was in a 5 minute Exercise Mode
2.	This is a new model (G0071760) that required changes to the evolution controller database.
        Could this change have affected some of the Modbus tables?
3.	“Additional Modbus” Timeout on the Advanced Page was set to 3.0 sec and "Modbus Exceptions" continued.
         Could be related to “Unknown Issue” #11
4.	“Modbus Exceptions” on Monitor page increases.by one every 8 minutes.
5.	Generator and Genmon program appears to work OK otherwise.

Environment
Generator Model: G0071760
Genmon Logs: have been Submit via the About page
Genmon Version: V1.16.09

### Screenshot or Pictures relating to the problem (if possible)
**Modbus Log while generator was in Exercise Mode.**
2021-05-24 10:44:18,659 : Error Receiving slave packet for register 0052
2021-05-24 10:52:32,281 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-24 10:52:32,283 : Error Receiving slave packet for register 0053
2021-05-24 11:00:45,894 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-24 11:00:45,896 : Error Receiving slave packet for register 0001
**--- 14 minutes….during exercise mode** 
2021-05-24 11:14:08,810 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-24 11:14:08,811 : Error Receiving slave packet for register 05f5

**Modbus Log from this morning..**
2021-05-27 07:15:35,828 : Error Receiving slave packet for register 05f1
2021-05-27 07:23:49,450 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 07:23:49,451 : Error Receiving slave packet for register 0033
2021-05-27 07:32:03,056 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 07:32:03,057 : Error Receiving slave packet for register 05f1
2021-05-27 07:40:16,669 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 07:40:16,670 : Error Receiving slave packet for register 005a
2021-05-27 07:48:30,290 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 07:48:30,292 : Error Receiving slave packet for register 001c
2021-05-27 07:56:43,921 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 07:56:43,924 : Error Receiving slave packet for register 005c
2021-05-27 08:04:57,559 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 08:04:57,562 : Error Receiving slave packet for register 000d
2021-05-27 08:13:11,167 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 08:13:11,168 : Error Receiving slave packet for register 000d
2021-05-27 08:21:24,789 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 08:21:24,791 : Error Receiving slave packet for register 0053

**“Monitor Page”
Communication Stats :**
Packet Count :          M: 1472690, S: 1472689
CRC Errors :                 0
CRC Percent Errors    : 0.00%
Timeout Errors :           0
Timeout Percent Errors : 0.00%
 **Modbus Exceptions : 96        increases slowly by one every 8 minutes**
 Validation Errors :      0
Sync Errors :                0
Invalid Data :              0
Discarded Bytes :       0
Comm Restarts :        0
Packets Per Second : 63.48
Average Transaction Time : 0.0307 sec

jgyates, 2021-05-27:
The firmware version is the same version that has been tested in the past (V1.11).  The thing that is odd is the 8 minutes part. This means that it is not a specific register causing the modbus exception (your logs back that up). Modbus exceptions are sent when the controller encounters something that it does not think is valid. This occurs on Evolution 2 firmware that uses "encrypted" modbus. A unique sequence is required to unlock the encryption. I have seen Generac use modbus exceptions on on other models  (industrial) when an invalid register requested. Normally, on Evo2, the modbus "unlock" sequence is sent every 3 seconds. We can try decreasing that value as it may be marginal on your generator model.

Try adding this value to /etc/genmon/genmon.conf

     evo2_unlock_interval = 2

If that does not work then try a value of 1 instead of two. You will need to restart genmon each time you change the value. You can restart on the advanced page. Let me know if that changes the behavior in any way.

yguy75, 2021-05-27:
evo2_unlock_interval is not in genmon.conf
Therefore I added to end of file
Saved copy of genmon conf
Sudo nano /etc/genmon/genmon.conf
evo2_unlock_interval = 2
Restarted via Advanced page.

**Received 3 modbus exception errors, then one every 8 minutes
Problem still exists.
Will now try evo2_unlock_interval = 1**

2021-05-27 10:08:21,839 : Error Receiving slave packet for register 0010
2021-05-27 10:16:35,453 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 10:16:35,456 : Error Receiving slave packet for register 05f1
2021-05-27 10:16:36,489 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 10:16:36,490 : Error Receiving slave packet for register 0009
2021-05-27 10:16:37,523 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 10:16:37,524 : Error Receiving slave packet for register 001f
2021-05-27 10:24:51,139 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 10:24:51,140 : Error Receiving slave packet for register 05f1
2021-05-27 10:33:04,775 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 10:33:04,777 : Error Receiving slave packet for register 020b

yguy75, 2021-05-27:
Did a Reboot via Advanced page.
**Problem seems to persist.**

11:00 - rebooted
11:06 - Got 1st set of 3 modbus exceptions (I think this is normal)
11:14 - after 8 minutes got modbus exception error.

Time for my walk and workout.
Will be back in about 2 hours.

**Last 4 lines in genmon.conf**
version = V1.16.09
model = G0071760
ignore_unknown = true
evo2_unlock_interval = 1

**Rebooted at 11:00**
2021-05-27 11:06:01,345 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 11:06:01,347 : Error Receiving slave packet for register 0053
2021-05-27 11:06:02,381 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 11:06:02,382 : Error Receiving slave packet for register 0052
2021-05-27 11:06:03,415 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 11:06:03,416 : Error Receiving slave packet for register 0001
2021-05-27 11:14:17,030 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-05-27 11:14:17,032 : Error Receiving slave packet for register 0212

jgyates, 2021-05-27:
I reviewed some logs from a 20 kw and 24 kw Evolution 2 and I see the same errors in the modbus log every 8 minutes. With that data point I am guessing that this is the same for all models with fw version 1.11. I don't recall this occurring with V1.10. I will look into this in more detail.

yguy75, 2021-05-27:
I don’t have a generator. I’m an 80 year old guy helping three young 70 year old guys install Genmon on their three generators. They all have the same. Model # G0071760.They are really impressed with Genmon’s cababilities. Thank you for all that you have done on this project.Don. Sent from Mail for Windows 10 From: jgyatesSent: Thursday, May 27, 2021 11:56 AMTo: jgyates/genmonCc: yguy75; AuthorSubject: Re: [jgyates/genmon] Modbus Exception and Error Receiving Slave Packet repeats every 8 minutes. (#539) I reviewed some logs from a 20 kw and 24 kw Evolution 2 and I see the same errors in the modbus log every 8 minutes. With that data point I am guessing that this is the same for all models with fw version 1.11. I don't recall this occurring with V1.10. I will look into this in more detail.

yguy75, 2021-05-27:
can you please delete the previous post.
Thanks.

yguy75, 2021-06-02:
Are there any consequences to running Genmon attached to a generator while still getting Modbus Exception errors every 8 minutes?

Are any calculations or values reported as incorrect?

liltux, 2021-06-02:
@yguy75 can you verify your unit controller, Evolution 2.0?, and can you verify the installed firmware version.  I would like to see if I can duplicate your Modbus Exceptions.

jgyates, 2021-06-02:
No. If a single modbus register request occurs every 8 minutes there would be no issue regarding the normal operation of genmon software. That particular register request would be skipped and read again in a few seconds. Genmon has a thread that does nothing but read all the registers over and over. If one read fails it uses the last good read value until it is read again. The logs indicate this failure is a different register every time it occurs so I can see any consequences other than your modbus.log file may fill up slightly faster than normal. I typically use a ram disk a ram disk for /var/log. Setting that up is described here:

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#extending-the-life-of-your-sd-card

liltux, 2021-06-02:
@jgyates looks like this is error occurs every 8 mins by the controller clock and not the Genmon software clock.  I would think this is a controller issue.  Maybe it is doing something internal every 8 minutes that disrupts the modbus?  just an observation.

yguy75, 2021-06-02:
Note: 
When the generator is exercising for 5 minutes, the modbus exception is 13 minutes, not 8 minutes.

yguy75, 2021-06-02:
From Maintenance Page
      Model : G0071760
      Generator Serial Number : xxxxxxxxxx
      Controller Detected :Evolution 2.0, Air Cooled

From About Page:
Genmon    Version V1.16.09

Re-submitting Logs

yguy75, 2021-06-02:
evo2_unlock_interval = 1

I haven't changed this parameter.
It is still in /etc/genmon/genmon.conf
Should I delete it?

yguy75, 2021-06-02:
Controller Hardware version 1.0

 Controller Firmware version 1.1

yguy75, 2021-06-02:
OPS..... sorry.... my mistake
This is the correct info.

Hardware Version : V1.00

Firmware Version :  V1.11

jgyates, 2021-06-03:
you should be able to delete the entry.

jgyates, 2021-06-11:
I added a note about this issue in the wiki under Known Issues (see item 15)

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

I am closing this for now. Let me know if this causes any issues for anyone. For now it looks like this is an acceptable issue as it does not appear to cause genmon to function incorrectly.

closed 2021-06-11 by jgyates

yguy75, 2021-07-01:
**There was No Modbus Exceptions while Generator was running on two occasions.**
Also there was a Phantom Outage for less then 1 sec.

**Outage.txt**

2021-06-17   12:29:44  ,0:00:00,  51.85 cubic feet
2021-06-24   00:09:34  ,1:35:07,  212.45 cubic feet
**2021-06-29   16:22:50  ,0:00:00,  270.30 cubic feet          Phantom Outage…. Received email**
2021-07-01   09:25:35,  1:20:57,  178.72 cubic feet  

Subject: Outage Recovery Notice at Jim's Generac
Time: 04:22:55 PM
Date: Tue 29-Jun-2021
Utility Power Restored. Duration of outage 0:00:00

**Modbus.log**
**First Outage**
2021-06-23 23:54:04,043 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-06-23 23:54:04,044 : Error Receiving slave packet for register 0036
2021-06-24 00:02:17,648 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-06-24 00:02:17,650 : Error Receiving slave packet for register 000f
2021-06-24 00:09:32,914 : Error: timeout receiving slave packet for register 041f Buffer: 0

  **OUTAGE -= No modbus exceptions for 1 hr 38 min.**

2021-06-24 01:47:01,973 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-06-24 01:47:01,974 : Error Receiving slave packet for register 0034
2021-06-24 01:55:15,597 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-06-24 01:55:15,598 : Error Receiving slave packet for register 0212

**Second Outage**
2021-07-01 09:14:40,076 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-07-01 09:14:40,077 : Error Receiving slave packet for register 0009
2021-07-01 09:22:53,690 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-07-01 09:22:53,691 : Error Receiving slave packet for register 000e

  **OUTAGE -= No modbus exceptions for 1 hr 31 min.**

2021-07-01 10:53:27,013 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-07-01 10:53:27,014 : Error Receiving slave packet for register 0055
2021-07-01 11:01:40,648 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-07-01 11:01:40,649 : Error Receiving slave packet for register 0020

I hope this additional information can help you.

jgyates, 2021-07-01:
Thanks for the data!