jgyates/genmon #540: Getting engine run time via MQTT

danodemano, 2021-05-27:
Trying to add run time onto my Home Assistant dashboard but I don't see it coming through the MQTT data stream anywhere. I can see the sensor in genmon and it appears to be correct. Can you please point me in the right direction on how to get this data? If it's not currently in the MQTT data stream then consider this a feature request to add it. 😁 Thanks!!

lmamakos, 2021-05-27:
It should show up in the `generator/Maintenance/Service/Total Run Hours` MQTT topic.  Check the MQTT extension configuration in genmon to ensure that you've not disabled that class of messages.

I don't believe the MQTT entity configuration that I post over on the Home Assistant community site actually picks out that value and exposing it in an entity.  I would be easy to add.

danodemano, 2021-05-27:
Great, thanks! I couldn't find a list anywhere of all the topics and I didn't see it in anyone's configs. It does show up there as expected though, thanks so much!!

closed 2021-05-27 by danodemano

lmamakos, 2021-05-28:
I added an entity to my Home Assistant configuration for this.  You can look at the sample configuration attached to the [first post on the Home Assistant community thread](https://community.home-assistant.io/t/monitor-your-generac-generator-with-home-assistant/62701).

danodemano, 2021-05-29:
Awesome, thanks so much!!