jgyates/genmon #541: Loss of communication after power outage

johnsdiyplayground, 2021-05-28:
If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Normal communication between generator and genmon.

### Actual Behavior

After 6 months of flawless operation, genmon stopped communication with generator following a power outage event that lasted approximately 1.5 hours.   System had been working great and taking on regular SW updates. 

### Steps to Reproduce (including precondition)

Have tried to reboot several times, power cycle, unplug / replug molex connection to generator.

### Screenshot or Pictures relating to the problem (if possible)

None.

### Your Environment

- Generator Model:  G0070420  22kW, Evolution (no wifi), s/n ...290005
- Genmon Logs:  Completed within the last 15 minutes
- Genmon Version:  1.16.09

jgyates, 2021-05-28:
your logs point to a break in your cable or bad RS-232 or a bad ground connection between the generator and the RS-232 interface. The software is not having any problems transmitting, yet it receives nothing back, so either the transmitted data is not being received or the controllers response is not being received. Review the steps on this page:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

Let me know how it goes.

johnsdiyplayground, 2021-05-29:
Problem solved - root cause is user error.  Found that I had the TX/RX pins on the Pi hooked to GPIO 10 and 12 instead of 8 and 10.   
 Must have reattached them incorrectly when inside the project box.  Works perfect again.  Thanks!

closed 2021-05-29 by johnsdiyplayground