jgyates/genmon #543: Time will not sync

restauffer, 2021-06-04:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Time will sync between Monitor and Generator when button is pushed in Maintanence screen

### Actual Behavior

When pushing "Set Generator Time" button it appears to be working and pops prompt that it may take up to one minute. However, time does not change. Time is off ~ 55 seconds

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment
[genmon.log](https://github.com/jgyates/genmon/files/6599935/genmon.log)
[genmon.log](https://github.com/jgyates/genmon/files/6599942/genmon.log)

- Generator Model:
- Genmon Logs: Submit via the About page
- Genmon Version: Version V1.16.09

jgyates, 2021-06-04:
I don't know of a way to set the seconds (only hours and minutes) via modbus. The software will wait for the seconds value for the system time to roll over to zeron on the pi, then it sets the time of the controller (this is why it can take up to a minute). This should result in a consistent time difference of one minute or less, which it looks like what you are seeing. Since most linux system use time servers to set the time this approach keeps the pi in sync with global time (at least within a few seconds if using NTP to set the time). In short if you are seeing a time difference between the generator and pi of less than one minute, the this is expected behavior.

Let me know if you have any other questions on this topic.

restauffer, 2021-06-04:
That makes perfect sense. I think 55 seconds will not make a difference in
anything but the peace of mind of an OCD person :)

Thanks for the quick response and easy-to-understand  explanation.

Ricke Stauffer
Email: ***@***.***
Voice: 734-426-4874
Cell: 734-216-1012

      “Pray, and let God worry”
Martin Luther

skipfire, 2021-06-04:
Actually 55 seconds might.  I thought about asking if anyone else had their generator always starting 1 minute after the set start time.  Instead of 10:00 for the weekly and 9:30 for the monthly I get 10:01 and 9:31 on the dots.

jgyates, 2021-06-06:
I am going to close this thread, feel free to post follow up questions or info here.

closed 2021-06-06 by jgyates