jgyates/genmon #545: hpanel+gts transfer switch - no OnUtilityChange notifications, transfer switch state: unknown

stranger2904, 2021-06-11:
Hi, Jgyates

First of all I would like to thank you for this amazing tool, it's working really well and as it's written in python - makes it quite easy to meet my needs  !

### Expected Behavior

As I'm poking around notifications and such, I noticed that I don't get onutility change notifications.
I have h_100 controller with a dumb GTS switch (which is recognized as 2-wire in system)
The system properly recognizing:

```
    Engine :
        Engine State : Stopped, Key SW Auto
        Generator Status : Stopped
    Line State :
        Transfer Switch State : Utility
```

But when I active test mode on the transfer switch, I see this:

```
    Engine :
        Engine State : Running from 2-wire
        Generator Status : Warmed up, Alarms On
    Line State :
        Transfer Switch State : Unknown
```

### Actual Behavior

I guess since after switchover to generator power initiated by dumb transfer switch - system can not tell position of the transfer switch and hence can not generate OnUtilityChange event, maybe add some logic/tweak to enable considering
        Transfer Switch State : Unknown as         Transfer Switch State : Generator  so appropriate notifications/logs could be generator

### Your Environment

- Generator Model: SG150/SG300
- Genmon Logs: Submitted alongside with registers
- Genmon Version: V1.16.09

  Controller Detected : H-100 Controller, PM-DCP Release 4.3 , HW00
    Controller Software Version : PM-DCP Release 4.3 , HW00, $Date: 2018/06/29 20:48:02 $
    Minimum GenLink Version : 4.10.0, 4.10.0

liltux, 2021-06-11:
@stranger2904 the H Panel uses a set of Aux Inputs to monitor switch position.  These inputs are located in the customer connections area of the generator.  When used with a GTS or other 2 wire start transfer switch, these inputs have to be wired up between the Transfer Switch and the Generator.  Typically using an aux contact off of the switch mechanism.

jgyates, 2021-06-11:
The software uses modbus register 0x0087 OUTPUT Register 6 bits D1 (0x0002) and D2 (0x0004) to determine the state of a "dumb" switch on H-Panel controllers. The code shows this here:

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_HPanel.py#L778

and here:

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_HPanel.py#L1188

As @liltux mentioned, your switch must be hooked to to these inputs so the H-Panel and genmon can report the state of the switch. If your switch is does not connect to these two inputs then the software does not know how to report these values.

stranger2904, 2021-06-12:
Ok, I think I found a missing connection from GTS transfer switch to the Generator Digital Input (GEN POWER DIGITAL INPUT - D14). 
Hopefully early next week I will be able to add that connection and see if it solves the issue (and post some pictures here)

stranger2904, 2021-06-15:
OK, so I was able to add missing wiring between GTS transfer switch and H-100 panel.
and now genmon correctly detects transfer switch state (utility/generator)
I still don't receive utility change notifications though, it's not super big deal as i can write my own script which will parse     
 "Transfer Switch State" output but maybe you have an idea on what it might be.
Also I noticed a bit weird thing (i was checking on generator panel itself):
When transfer switch on the city power - I have status "1" on digital input 7 and "0" on digital input 8
When transfer switch on the generator power - I have status "1" on digital input 7 and "1" on digital input 8

skipfire, 2021-06-16:
@stranger2904 was the generator running when you saw the digital input 7 high?  7 is pin 26 which is generator running, so nothing to do with the transfer switch.  IO 8 is pin 24, which maps to exercising, which if the power is on I'm guessing that means exercising with load.  GenMon (at least on mine with Evo2 controller) can detect utility state completely apart from transfer switch state.

stranger2904, 2021-06-16:
@skipfire yes, it was still running but it didn't changed once gen was down.
According to the H-100 manual:
[image]
I don't think behaviour is the same between evo2 and h-100.

skipfire, 2021-06-16:
Sorry, I thought you meant IOs on the Pi.  If the city power was never down then it makes sense that pin 7 was always high, the generator has to know when utility power is present to know if it can switch back and not just rely on the transfer switch state.  I believe pin 8 should only be high when both the transfer switch is on generator and the generator is running.

jgyates, 2021-06-16:
Presently genmon does not notify of an outage for the industrial units. This is due to the wide variety of transfer switches and the variety of ways to connect them. I tried to support this but there were quite a few instances like this one where either the connections were not consistent across installation or a different switch was used. I could try to add this however my fear is that there would be a lot of folks getting notification of outages when there are none.

stranger2904, 2021-06-16:
Got it, @jgyates no worries, as long as it's expected behaviour that's totally cool !
If you can add some additional logic which could be enabled on-demand that would be amazing .
But I'm pretty confident that I can modify code/add another module to work for our setup myself.

jgyates, 2021-06-16:
Let me look into that. The current code that detects outage relies on knowing the utility voltage. On industrial models you either have knowledge of the switch state (utility/generator) if wired correctly, or if you are using a smart switch then you get utility voltages so the code would need to be modified somewhat to make this happen.

stranger2904, 2021-06-16:
Ok, thank you, let me know if you would like me to test anything or there is any additional information/data required.

jgyates, 2021-06-17:
I checked in V1.16.11. For H and G panel systems, this update adds a new option on the settings page named "Outage Notice on Transfer State Change". If enabled this will look at the state of output register 6 on H Panel and output register 5 on G panel. If the registers indicate the transfer switch is in the "generator" position then and outage notice will be sent, A utility notice will be sent if the output register indicate the switch is in the "utility" position. This option is off by default as this depends on 

      1) The switch is connected to the controller properly
      2) The assumption is made that the transfer switch position is an indication of the outage state. Without a HTS type switch there is not way to read the utility voltage on H and G panel controllers.

I am closing this thread for now but please let me know the status of your testing by posting to this thread. I did test with a emulator.

closed 2021-06-17 by jgyates