jgyates/genmon #546: Possible Gengpio.py Conflict

Clarke219, 2021-06-11:
I am still in the process of getting everything set up. I enabled Gengpio.py as I want to connect leds for the Ready and Alarm status to the outside of my enclosure box. Since enabling it, the Pi shuts down after what appears to be enough time to boot and run Genmon. I have a switch and appropriate code to shutdown the Pi by shorting pins 5 and 6. The software waits for a change in state of pin 5 using wait_for_edge [GPIO.wait_for_edge(3, GPIO.FALLING)] and issues a shutdown command when it happens. 

Since Gengpio.py also uses pin 5 for the internet status, I'm wondering if the initialization of pin 5 by Gengpio.py [GPIO.setup(ER_INTERNET, GPIO.OUT, initial=GPIO.LOW)] is triggering the shutdown? If this is true, since I have Genmon run at startup, I appear to be caught in an endless loop.

Do you have a suggestion on how to resolve it without taking the card out of the Pi? And, once it is fixed, can I change the assignment of ER_INTERNET to a different pin? Does Gengpio,py get updated so that any reassignment that I made would just get overwritten?

Thank you for any help that you can provide.
Clarke219

jgyates, 2021-06-11:
I added a new config file for gengpio.py that will allow you to change the default pin number with V1.16.10. The install program will add the new conf file to /etc/genmon on new installs. If you already have this installed then genmon will copy the file to /etc/genmon once you enable the gengpio add on. Since you have another program that is conflicting you should disable genmon at boot or the other program that is conflicting, then you can copy over the file from /home/pi/genmon/conf/gengpio.conf to /etc/genmon

     sudo cp /home/pi/genmon/conf/gengpio.conf  /etc/genmon

Then you can edit the file /etc/genmon/gengpio.conf to change the pin numbers.

Note that the conf file sets the pin number, not the GPIO number.

Let me know if you have any questions.

Jason

Clarke219, 2021-06-13:
Jason-

Thank you for your reply. I have copied and edited the gengpio.conf. Can you tell me how to disable genmon at boot? And then enable it again when everything is working?

Thank you,
Clarke219

jgyates, 2021-06-13:
Sure. If you ssh into your pi, from the command line type:

     sudo crontab -e 

This will allow you to edit the cron script for the root user. You may be prompted to select and editor when you type the above command, if you do select nano as the editor.

Once the cron script is visible, you should see a line like this:

          @reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start

If you add a pound sign (#) at the beginning of this line, then save the file by typing Ctrl+x, then Y then Enter.

This will disable genmon from loading on the next boot. To re-enable repeat the steps and remove the pound symbol at the beginning of the line.

Clarke219, 2021-06-13:
Once again, Thank You!

jgyates, 2021-06-14:
I am going to close this issue but feel free to post to this thread if you have any comments for questions on this topic.

closed 2021-06-14 by jgyates