jgyates/genmon #548: Modification to allow a two propane tank gauges with [gentankdiy]

speters0, 2021-06-18:
Hi,

I'm living in the northern Laurentians Quebec, Canada where temperature can get as low as -40°C
I have a 16KW Generac generator and a 5 burner Cooktop. They all run on propane gas.
I have two 420lb propane tanks. And the reason why I have two tanks is:
If It gets too cold the propane gas might freeze if I have only one tank.
If the demand on pressurized gas is high, the pressure drop at the outlet of the tank regulator will lower the temperature substantially at that point.
It is the thermodynamic law.
By having two tanks, the pressure drop at each outlet is less and it minimize the chance of gas freeze.
Lot of peoples on northern locations uses at least a two tank setup.

I'm using DIY thank Method 1 and adding two Propane Fuel Gauge (on on each tank).
The liquid propane level on each tank can lower at a different rate depending on brass fitting diameter and the length of the gas line.

So I decided to modify Genmon Program to accommodate for a two Gauge setup.

The is a new option in the setup file: gentankdiy.conf

# number of tanks (1 or 2)
nb_tanks = 1

If the value is 2

There is a new External Tank Gauge that will display on the [Status] page of Genmon.
In the [Maintenance] page, the fields (Fuel Level Sensor) and (Fuel in tank (sensor)) are calculated with the mean value of both gauges.

The modification has been made on Version V1.16.09

It is only applied on Evolution Controller

Here is the list of the modified files:

/home/pi/genmon/conf/gentankdiy.conf
/home/pi/genmon/genmonlib/gaugediy.py
/home/pi/genmon/gentankdiy.py
/home/pi/genmon/genmonlib/controller.py
/home/pi/genmon/genmonlib/generac_evolution.py
[Genmon.zip](https://github.com/jgyates/genmon/files/6679090/Genmon.zip)

If you wish it could be interesting to have a look at the code, adjust it if needed, and merge it to the latest version.⍺

I will upload these files on github site.

Thank you!

jgyates, 2021-06-19:
Sounds like a useful addition. Let me review the code and look at integrating this feature.

jgyates, 2021-07-06:
I integrated most of your changes, however I had to make some tweaks so there is not impact to people not using two tanks. In short you just need to add 

     nb_tanks = 2 

to /etc/genmon/gentankdiy.conf and restart genmon. A refresh of the browser would need to be done if the browser is open while genmon restarted.

Two gauges are shown on the main page if this is enabled. Note: this only applies to type 1 DIY gauges. Type 2 DIY gauge only supports one instance at the moment.

Let me know if you have any issues.

closed 2021-07-06 by jgyates

jgyates, 2022-04-11:
That is do-able. I will try to do that soon. I will update this thread when complete. Let me know once you have a chance to test this as I do not have three tanks to test with :)

jgyates, 2022-04-11:
@davisgoodman, just to clarify, are you OK with creating your own gauges with the method in this wiki entry:

https://github.com/jgyates/genmon/wiki/Appendix-L-Adding-a-Propane-Fuel-Gauge-to-Genmon#diy-tank-method-1

See DIY tanks gauge method one. This is the only type of gauge that currently supports multiple instances.

davisgoodman, 2022-04-11:
Yes, does either of these two are supported or only the one with the R3D Rochester sensors. Either or I'm ok to create them

jgyates, 2022-04-11:
OK. The DIY Tank Method 1 is the only one that supports multiple tanks.

davisgoodman, 2022-04-11:
Ok. As soon as I can put my hand on those sensors, I’ll update the thread.

jgyates, 2022-04-11:
@davisgoodman 

I have added support for up to 4 tanks (the max supported by the design of the circuit). These updates are checked in but I have not increased the version number yet. Let me know once you have an opportunity to test.

davisgoodman, 2022-04-11:
Wow..  Thanks. As soon as I can put my hands on those sensors I sure will.

I'll reach out to speter0 to find out where he got his.

speters0, 2022-04-11:
Hi,

I bought the gauges at:
https://www.petroviron.ca/

davisgoodman, 2022-04-11:
Thanks, that’s actually one of the distributors they gave me.

 

I’ll phone them up.

 

https://www.petroviron.ca/

davisgoodman, 2022-04-12:
@speters0Thanks for the lead, I have 4 sensors on their way.

davisgoodman, 2022-04-14:
Still wating to get my sensors but updated genmon on my RaspPI.

 

Maybe just something that was missed but the gentankdiy.conf still shows:

# Optionally support two tanks of gauge type 1, the default is 1.

# Only valid data is 1 or 2

nb_tanks = 1

 

I guess only the comment needs to reflect the update

jgyates, 2022-04-14:
The file /etc/genmon/gentankdiy.conf does not get updated however the file in /home/pi/genmon/conf/gentankdiy.conf does get updated. If the files in /etc/genmon were updated every update it would wipe out your site specific settings.

davisgoodman, 2022-04-14:
Indeed, didn't realize it.

speters0, 2022-04-15:
Tank gauges still working great with the last mod!

davisgoodman, 2022-04-15:
Glad to hear! 

 

Can’t wait to get my 3 tanks hooked up.

davisgoodman, 2022-04-18:
Well  got my gauges today and all installed.
[image]

Few questions, 
when 3 gauges are installed is the "20%" averaged across all 3 gauges?
Do they need to be calibrated?
Is there a config for setting the threshold (20%)?
Is there a way to send out only "Low Fuel" emails as I  may want to send the emails directly to my Propane provider but wouldn't want have any other kind of notifications from Genmon?

davisgoodman, 2022-04-18:
Also would there be a way to export the tank readings through MQTT, would allow to integrate them with Home Assistant

jgyates, 2022-04-18:
Genmon will send email notices at 10% and 20% fuel levels.  Short message notices for add ons (SMS, etc) are sent at 20%. When multiple tanks are configured this is the average of the total tanks.

I will leave the calibration question to others as I do not know the answer. For type 1 gauges, i assume no but I do not know for sure.

There is not a configuration setting for fuel level percentage notices.

Low Fuel messages are sent in the "warn" category on the Notifications page. and in "Warning" category for short messages for Add ons. Other types of messages are also in the warning category.

I can look into the MQTT export.

davisgoodman, 2022-04-18:
Thanks for the response. I suppose there are other kind of warning that would go out in this category which I wouldn't want to send them. For now I'll notify them when I get it.

jgyates, 2022-04-18:
I checked in a mod to show readings of each tank on the maintenance page and they will also be exported via MQTT.

davisgoodman, 2022-04-19:
Upgraded to the latest. On the maintenance page I only get the average as well as in MQTT
[image]
[image]

jgyates, 2022-04-19:
any errors in /var/log/genmon.log?

davisgoodman, 2022-04-19:
> any errors in /var/log/genmon.log?

Nothing particular:
2022-04-18 18:33:11,945 : GenMon Loaded for site: 105 Des Gardénias, Val-David using python 3.9
2022-04-18 19:46:47,400 : Generator Monitor Shutdown
2022-04-18 19:46:50,903 : Selected Controller: generac_evo_nexus
2022-04-18 19:46:52,093 : GenMon Loaded for site: 105 Des Gardénias, Val-David using python 3.9
2022-04-18 20:10:20,725 : Generator Monitor Shutdown
2022-04-18 20:10:23,570 : Selected Controller: generac_evo_nexus
2022-04-18 20:10:24,767 : GenMon Loaded for site: 105 Des Gardénias, Val-David using python 3.9

jgyates, 2022-04-19:
send your logs from the about page. This requires outbound email to be working. If you don't have outbound email let me know and we can to this a different way.

davisgoodman, 2022-04-19:
Done!

jgyates, 2022-04-19:
try this: 

edit the file /etc/genmon/gentankdiy.conf

add the line

debug = True 

to the end of the file, save the file, restart genmon, wait 20 seconds, then send me the logs again

I don't have the hardware on the i2c but if you enable "debug" in the gentankdiy.conf file it will ignore fatal errors and output more info to the log. The gauges data is displaying for me with debug enabled.

jgyates, 2022-04-19:
Try updating the repository again. I think I forgot to push the update. Hopefully that corrects the issue. Let me know

jgyates, 2022-04-19:
also, if this works take out the debug=True from the /etc/genmon/gentankdiy.conf file as that extra log info can put unneeded wear on your SD card.

davisgoodman, 2022-04-19:
:-( 
[image]

Still....

davisgoodman, 2022-04-19:
Just submitted the logs .....

jgyates, 2022-04-19:
update to the latest, then send logs

davisgoodman, 2022-04-19:
Upgraded and sent logs. Still only seeing the average

jgyates, 2022-04-19:
it looks like github is not updating the repository at your location or for some other reason you are not getting the update. I put some additional log output in the latest update but your logs do not show this info.

For example link 92 of ./genmon/gentankdiy.py should be 

     self.LogDebug("Num Tanks: " + str(self.nb_tanks))

and I am showing this in the repository but I would guess that your pi does not have this update. 

https://github.com/jgyates/genmon/blob/f1a01f7af3433bf7cbda5c39d4ac71042a01d4e2/gentankdiy.py#L92

davisgoodman, 2022-04-19:
Do test I create a separate directory and when trying to clone the repo I'm getting this:
pi@generac:~/gitdownload $ git clone https://github.com/jgyates/genmon.git
Cloning into 'genmon'...
error: git-remote-https died of signal 4

davisgoodman, 2022-04-19:
Ok so the issue is definitely the update. 

For testing I copy/pasted gaugediy.py from the repo into a new file and diff and the one on my side missed 12 lines of codes

jgyates, 2022-04-19:
you can also manually update with this command from the genmon folder.

   ./genmonmain.sh -u

This ends up calling these commands:

      git fetch origin
      git reset --hard origin/master

If you have some sort of connectivity issue with github this would cause the problem we are seeing.

jgyates, 2022-04-19:
when you select update form the web interface it ends up calling genmonmain.sh

davisgoodman, 2022-04-19:
So you're right, some issues with GitHub.

pi@generac:~/genmon $ ./genmonmaint.sh -u
This script will update genmon to the latest version from the github repository. This script requires internet access. If you have modified any files in the genmon directory, they will be overwritten. Configuration files in the configuration directory will not be overritten.  Continue? (y/n)  y
Updating genmon...
error: git-remote-https died of signal 4
HEAD is now at 02f447e V1.18.10
Done.

davisgoodman, 2022-04-19:
Just let me know which files been mofidied, I'll pickup only those ones and replace them

davisgoodman, 2022-04-19:
I believe gaugesdiy.py, controller.py, tooltips.txt, gettankdiy.py

jgyates, 2022-04-19:
/genmon/gentankdiy.py
/genmon/genmonlib/controller.py
/genmon/genmonlib/gaugediy.py

jgyates, 2022-04-19:
toottips.txt also, but that one is not critical, just some text for the UI help bubbles related to multiple tanks.

davisgoodman, 2022-04-19:
Well I now see all three gauges in the Maintenance, though MQTT has lost the Maintenance export
[image]
[image]

jgyates, 2022-04-19:
send logs

davisgoodman, 2022-04-19:
And something doesn't jive in the numbers. Fuel Level Sensors seems to be the average which makes sense but then we should have 3 more readings 1 for each tank. Or have only 3 readings 1 per tank.

Just submitted the logs

jgyates, 2022-04-19:
Are these logs that have run **after** you requested data from MQTT? That would be more helpful

davisgoodman, 2022-04-19:
Yes

davisgoodman, 2022-04-19:
Just as an FYI, I  went elsewhere in github and was able to clone a few other repos without issues. I tried your other top repo githup.io and had the same issue/error

jgyates, 2022-04-19:
I checked in an update to genserv.py and controller.py try that. If you have the setting "JSON for Numerics" enabled it would caused an issue.

Regarding the repo access issue. I don't have any control on my end about access to the repository. I would not know where to begin troubleshooting that issue. Can you duplicate this on another computer? or is it specific to your pi?

This url is what the web interface uses:

     http://192.168.1.10:8000/cmd/maint_json

the mqtt plug in uses the above or this command, depending on the "JSON for Numerics" setting:
 
     http://192.168.1.10:8000/cmd/maint_num_json

davisgoodman, 2022-04-19:
Ok so the issue seems to be only on the Pi. I'll look into this later. As for the update, everything is good now. Both the maintenance page and MQTT shows the 3 tanks reading.

Thanks for all the effort

davisgoodman, 2022-04-19:
If I can make a small comment and that is only a matter of clarity I would probably have the label "Fuel Sensor" changed to Fuel Sensor Tank 1" to be consistent with the other 2 labels.

jgyates, 2022-04-19:
Thanks for the feedback. I choose the first tank to be labeled "Fuel Sensor" and not "Fuel Sensor Tank 1" because the vast majority of uses only have one tank and I did not want to confuse users. Also, adding the additional logic to make this display differently for multi tank installations vs single tank installations would introduce maintenance complexity and make the first tank display differently via MQTT topics. I did consider this but I opted to implement it this way.

davisgoodman, 2022-04-19:
Totally understand. But now I guess with multi-tanks we don't have an average reading anymore. I guess that is the consequences of having all 3 readings now.

Again thanks a lot for implementing this in no time.

davisgoodman, 2022-04-20:
Hi Jason, just found out that the vendor had sent me the datasheet of the Rochester gauges sensors. I was just wondering if the calculation  done for the percentages are based on these.
[image]

jgyates, 2022-04-20:
Hi @davisgoodman 

I don't know. I did not create the DIY method 1, @zekyl314 created this. Hopefully he can weigh in.

zekyl314, 2022-04-20:
Hi @davisgoodman, as far as I recall, the sensor does read up to 5V at 100% full.  I pulled some of the code and details from another generic project for reading the hall effect sensors.  The genmon configuration file shows the step of 20% per volt, which would match up to their documentation and my knowledge. 

I have confirmed a matching reading on my physical gauge vs the digital gauge reading in genmon.  The digital is more precise than the physical and both change a little based on temperature.  

I hope this helps.

davisgoodman, 2022-04-20:
That's perfect, thanks for confirming. I figured it had been done and verified before.

zekyl314, 2022-04-20:
You're welcome @davisgoodman!  Love hearing people using this addon.  :)

[41 more replies]