jgyates/genmon #549: GPIO uses?

skipfire, 2021-06-20:
Are there things in GenMon that use GPIO other than gengpio.py and gengpioin.py?  It looks like the only pins not in other use are 7, 12, 19, 23, 27, and 28.

jgyates, 2021-06-20:
There are two groups of programs: genmon / genserv and the collection of add on programs.

genmon / genserv only use serial Tx and Rx which share the pins for GPIO 14 and 15 (pins 8 and 10). If you use serial over IP then these are not used by genmon.

for the add on program, it depends on which one you have enabled, if you are just using gengpio and gengpioin then the ones listed in gengpio.conf are used buy gengpio and pins 11, 13 and 15 are used if using gengpioin.  

gentankdiy uses I2C which shares pins 3 and 5 (GPIO 2 and 3). However this conflicts with gengpio as it uses 3 and 5 by default but you can change that via gengpio.conf

skipfire, 2021-06-20:
Thanks, I'm using 4 pins on a HAT design and wanted to try to minimize potential overlap.

closed 2021-06-20 by skipfire