jgyates/genmon #550: Suggestion for printer-friendly Service Journal page

johnsdiyplayground, 2021-06-21:
Hi Jason!  Your GenMon software is excellent, have been using it for about a year now.  I have one feature suggestion, which is to have a click button or some other method to get a printer friendly version of the Service Journal page.  I'm an older guy so I still like to see stuff in print, and would throw a copy into my Generac file folder at least once a year.  

The software as of today creates a page that is too wide to print and cuts off the right side of the comments, in portrait or landscape mode.  I have tested this is Brave, Chrome, and Edge browsers with similar results.

jgyates, 2021-06-22:
Hi @johnsdiyplayground 

Sounds like a good idea. I will try to work this in.

Thanks
Jason

jgyates, 2021-07-06:
FYI, another thing you can do is convert the raw service log to a CSV format then import it to a spread sheet.

If you input the contents of the file /etc/genmon/maintlog.json into this site: https://konklone.io/json/
 then you can download a CSV (comma separated value) file into a spreadsheet.

johnsdiyplayground, 2021-07-06:
Hi Jason, that is a cool trick!  I tried the export out as you suggested
and was able to get a nice printable spreadsheet in LibreOffice Calc.
Thanks for the tip!

jgyates, 2021-07-07:
I have added the printer friendly feature to the list of features to be added:

     https://github.com/jgyates/genmon/blob/master/changelog.md

I will update this thread once that feature has been added to the code. For now I will close this issue but feel free to post here if you have questions.

jgyates, 2021-07-07:
It is actually this link:

     https://github.com/jgyates/genmon/blob/master/feature_requests.md

closed 2021-07-07 by jgyates

jgyates, 2021-07-10:
V1.16.13  now supports a print button on the service log page.

johnsdiyplayground, 2021-07-10:
Thanks Jason!  I got the update installed and the print button on the
service log page works great!  I am sending you another donation for your
work, I really appreciate it.