jgyates/genmon #551: Generac Evolution Firmware Changed from V1.11 to Vx.xx and back in 5 seconds.

yguy75, 2021-06-26:
Generac Evolution 2.0 Firmware Changed from V1.11 to Vx.xx and back in 5 seconds.

The first time this message showed up was after updating to Genmon V1.16.10 on June 11.     

Issue  #546  may be of interest.
Yates said “added a new config file for gengpio.py that will allow you to change the default pin number with V1.16.10.”

It appears to be some sort of phantom message as it corrects itself 5 seconds later.

Warnings are very sporadic…. 
No regular interval or pattern.

1st warning  - 16 hours after V1.16.10 installed
2nd warning – 23 hours later 
3rd warning  - 76 hours later   (3 days, 4 hours)
4th  warning  - 5 hours after updating to V1.16.11
                And  117 hours after 3rd warning (4 days, 21 hours) 

No warning in the last 5 days….
Interval seems to be getting progressively longer??

**Received email warnings for each occurrence**.
NOTE: This message is a notice that the version of the generator controller
firmware has changed from V1.11 to V0.32

**From Genmon Log**

**2021-06-11 17:24:35,113 : New version detected: Old = V1.16.09, New = V1.16.10**   
2021-06-11 17:24:35,410 : GenMon Loaded for site: Jim's Generac using python 2.7
**2021-06-12 09:21:07,784 : Frimware Changed from V1.11 to V0.32  
2021-06-12 09:21:12,804 : Frimware Changed from V0.32 to V1.11**
2021-06-12 11:12:20,247 : Generator Monitor Shutdown
2021-06-12 11:12:23,121 : GenMon Loaded for site: Jim's Generac using python 2.7
2021-06-12 12:09:50,381 : Generator Monitor Shutdown
2021-06-12 12:09:53,288 : GenMon Loaded for site: Jim's Generac using python 2.7
2021-06-12 15:36:30,805 : Generator Monitor Shutdown
2021-06-12 15:36:33,666 : GenMon Loaded for site: Jim's Generac using python 2.7
2021-06-13 06:36:58,588 : Generator Monitor Shutdown
2021-06-13 06:37:01,459 : GenMon Loaded for site: Jim's Generac using python 2.7
**2021-06-13 08:45:16,020 : Frimware Changed from V1.11 to V0.00                               
2021-06-13 08:45:21,042 : Frimware Changed from V0.00 to V1.11**
2021-06-13 12:38:59,090 : Generator Monitor Shutdown
2021-06-13 12:39:48,592 : GenMon Loaded for site: Jim's Generac using python 2.7
2021-06-13 18:57:18,748 : Generator Monitor Shutdown
2021-06-13 18:57:21,476 : GenMon Loaded for site: Jim's Generac using python 2.7
2021-06-13 20:32:16,677 : Generator Monitor Shutdown
2021-06-13 20:32:19,467 : GenMon Loaded for site: Jim's Generac using python 2.7
2021-06-15 20:40:33,240 : Generator Monitor Shutdown
2021-06-15 20:40:36,093 : GenMon Loaded for site: Jim's Generac using python 2.7
2021-06-15 20:48:26,063 : Generator Monitor Shutdown
2021-06-15 20:48:28,972 : GenMon Loaded for site: Jim's Generac using python 2.7
**2021-06-16 12:03:12,157 : Frimware Changed from V1.11 to V0.32
2021-06-16 12:03:17,925 : Frimware Changed from V0.32 to V1.11**
2021-06-16 21:59:20,905 : Generator Monitor Shutdown
2021-06-16 21:59:23,903 : New version detected: Old = V1.16.10, New = V1.16.11
2021-06-16 21:59:24,189 : GenMon Loaded for site: Jim's Generac using python 2.7
**2021-06-21 09:52:19,782 : Frimware Changed from V1.11 to V0.32
2021-06-21 09:52:25,551 : Frimware Changed from V0.32 to V1.11**
2021-06-21 17:13:38,016 : Generator Monitor Shutdown
2021-06-21 17:13:40,900 : GenMon Loaded for site: Jim's Generac using python 2.7
2021-06-22 09:37:44,979 : Generator Monitor Shutdown
2021-06-22 09:37:47,792 : GenMon Loaded for site: Jim's Generac using python 2.7
2021-06-22 09:44:22,582 : Generator Monitor Shutdown
2021-06-22 09:44:25,488 : GenMon Loaded for site: Jim's Generac using python 2.7
2021-06-22 09:46:14,094 : Generator Monitor Shutdown
2021-06-22 09:46:16,891 : GenMon Loaded for site: Jim's Generac using python 2.7

      Model : G0071760
      Generator Serial Number : xxxxxxxxxx
      Controller Detected :Evolution 2.0, Air Cooled

Genmon    Version  V1.16.11

Hardware Version : V1.00

Firmware Version : V1.11

-
- Genmon Logs: Submit via the About page

jgyates, 2021-06-26:
This notice is issued when the modbus register that contains the firmware version changes. If you have intermittent communication errors this could be the source of the problem. What I would suggest is this: the next time you see this error, look at the communication stats on the monitor page. If you see any type of communication error then this is likely the problem. We can troubleshoot this if this is the case. The Evolution 2 will generate a modbus exception every 8 minutes, but this is expected. A timeout error could cause this condition. If that is the case then you could extend the timeout of modbus communications on the advanced page.

I got your logs, but your modbus log is filled with the evolution 2 modbus exception every 8 minutes so I can't see far enough back to see if there are other comm errors causing this since the logs are rotated to keep the file sizes small.

yguy75, 2021-06-27:
Thank you for your quick response.
We will monitor for communication errors and extend the timeout if required.

yguy75, 2021-06-27:
This is a new warning and the first time this message showed up was after updating to Genmon V1.16.10 on June 11, and it continued after V1.16.11.
Nothing else has changed.

jgyates, 2021-06-27:
Yes, however nothing related to serial communications has changed in quite some time. The only thing that I can think of that would cause this error is the modbus comms getting out of sync.  A timeout error would do this. Also, I am not getting reports of this error from others so I am assuming this is specific to your system.

yguy75, 2021-06-27:
Thanks again.
We’ll start by reviewing the wiring.

jgyates, 2021-06-29:
In the next release I will include a change that will only check for a firmware update every hour. Presently gemon reads a list of modbus registers over and over. Once the list has been read it will check for a firmware update, then read the list again and recheck, so this firmware check is done many times in an hour. With the change the chance that a modbus timeout will cause this error is greatly reduced. I will post here once the update is checked in a tested.

yguy75, 2021-06-29:
There have been no warning messages in the last 8 days.
The warning seemed to start after a version change or restart.
I will do a cold restart and monitor for a few days.
There have been 6 communication timeouts in last 8 days.

    Communication Stats :
        Packet Count : M: 19103780, S: 19103773
        CRC Errors : 0 
        CRC Percent Errors : 0.00%
        Timeout Errors : 6
        Timeout Percent Errors : 0.00%
        Modbus Exceptions : 1211
        Validation Errors : 0
        Sync Errors : 0
        Invalid Data : 0
        Discarded Bytes : 0
        Comm Restarts : 0
        Packets Per Second : 63.39
        Average Transaction Time : 0.0307 sec

yguy75, 2021-07-01:
After a cold start, there were no strange readings.

**BUT.... after a Restart Genmon Software on the Advanced page, we received very high Utility Max Voltage.**

Utility Max Voltage readings of 8224 V, or 13880 V, or 16189 V.

Received three emails with a change of state each time there was a Restart Genmon Software

NOTE: This message is a notice that the state of the generator has changed.
The system is not in alarm.
    Line : 
        Utility Voltage : 243 V
        **Utility Max Voltage : 8224 V**
        Utility Min Voltage : 236 V
        Utility Threshold Voltage : 156 V

Line : 
        Utility Voltage : 243 V
        **Utility Max Voltage : 16189 V**
        Utility Min Voltage : 235 V
        Utility Threshold Voltage : 156 V

    Line : 
        Utility Voltage : 243 V
        **Utility Max Voltage : 13880 V**
        Utility Min Voltage : 237 V
        Utility Threshold Voltage : 156 V

**After a Reboot System from the Advanced Page, readings seem OK.**

Also Register "023A" has the value 13880, even when Status Page shows correct voltages.

Line :
        Utility Voltage : 240 V
        Utility Max Voltage : 248 V
        Utility Min Voltage : 197 V
        Utility Threshold Voltage : 156 V

Resubmitted Registers

jgyates, 2021-07-01:
You have communication errors (or you are experiencing a very odd controller problem). This is multiple instances of invalid modbus data (firmware version and now utility voltage). 

This core issue is that the modbus protocol does not have a method to cancel a request so in the case where a packet times out (i.e. does not get a response back) the next packet is requested. If the controller responds with the first (timed out) packet, then invalid data is reported. This is a limitation of the modbus protocol. 

The ideal solution is to determine the source of the communication errors (e.g. faulty cable, bad ground connection, etc). If you are using a direct serial connection at a cable length within the RS-232 specification you should not get any timeout errors.

yguy75, 2021-07-01:
Thanks for your analysis and recommendations.

We will double check all of our wiring and get back to you.

jgyates, 2021-07-06:
I am going to close this issue for now but feel free to comment here if you have additional data. The next release (V1.16.12) will reduce the frequency that the firmware version is checked.

closed 2021-07-06 by jgyates

yguy75, 2021-07-06:
**Thank you for reminding us to recheck the wiring.**
We redid the wiring on the Pi side and on the Generator side.
Genmon has been now running for 27 hours with no errors.
I never realized how many different errors could be caused by a slightly loose connection.

BUT... at 11:00 yesterday when the generator was in exercise mode, we got a **timeout error** and an error in the mymodbus.log which was in between the normal 8 minute modbus exception. 
Also, the normal 8 minute error was increase by the 5 minute exercise period to 13 minutes.

2021-07-05 10:57:39,049 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-07-05 10:57:39,050 : Error Receiving slave packet for register 0258
**2021-07-05 11:00:15,182 : Error: timeout receiving slave packet for register 0009 Buffer: 0**
2021-07-05 11:11:01,678 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-07-05 11:11:01,679 : Error Receiving slave packet for register 0001

Other then this everything seems OK.

We will now update to V1.16.12 and continue monitoring,