jgyates/genmon #552: Possibility of connecting Genmon to Homebridge (and ultimately Homekit)

muddycharles68, 2021-06-29:
Hi there - I love Genmon, and have been using it now for over a year. Living in florida its really useful to be able to monitor the generator remotely.
Being a Mac / IOS household with Homebridge installed to bridge Somfy Shades, Insteon Switches and Nest thermostats into Homekit, it left me wondering if you have thought about this for Genmon?

jgyates, 2021-06-29:
I would guess that there is some method out there now that will allow homebridge to connect to a MQTT server (if homekit does not support that already). I will try to look into this as I am not that familiar with interfacing to homebridge / homekit.

muddycharles68, 2021-06-29:
OK thanks - looking forward to hearing what you find out!

muddycharles68, 2021-06-29:
FYI - not sure how to decipher this but it might be what you are looking for
https://www.npmjs.com/package/homebridge-mqtt
It does seem to be available?

lmamakos, 2021-07-02:
There has been a bunch of work integrating genmon with Home Assistant using MQTT.  However, I don't think that it is directly in a form that's compatible with the specific scheme outlined in that package that was referenced.  I've done some work on exposing genmon to Home Assistant, can you can read some of that journey at https://community.home-assistant.io/t/monitor-your-generac-generator-with-home-assistant/62701 which sort of presumes some familiarity with Home Assistant, but might give you a sense of things.

You could adopt Home Assistant as the center of your home automation system.  It integrates with existing Home Kit compatible peripheral devices.  I do this with a Home Kit compatible window AC unit that Home Assistant discovered, much to my surprise :-)

If HomeKit is the center of your universe, I believe that Home Assistant could export devices that it manages into a Home Kit installation.  I've not actually done this myself.  You could install a tiny Home Assistant instance just for this purpose.. You could take the Home Assistant entities that correspond to data exported from genmon, and expose those to HomeKit.  https://www.home-assistant.io/integrations/homekit/ talks about this particular use-case.  It sort of assumes that you already know about Home Assistant, but maybe the terms used there will be meaningful to you.

I hesitate to offer much specific device since I know only enough about HomeKit to be a danger to myself and others :-)

jgyates, 2021-07-31:
I have added this request to the feature list: https://github.com/jgyates/genmon/blob/master/feature_requests.md

If this gets added in the future I will update this thread.

closed 2021-07-31 by jgyates