jgyates/genmon #553: What information would be required to add support for a Cummins Quiet Connect generator?

derekatkins, 2021-07-02:
I moved, and I'm looking to get a generator at my new house.  I'm debating between a Generac Protector QS-38 and Cummins RS40.  The installer I used last time for my 22kW Generac has changed over from Generac to Cummins.  I trust their judgement, and it sounds like the RS40 is a better built system than the QS-38 (even if it'll be about $2k more).

Having said that, I'd like to re-introduce my genmon system to monitor the generator.  If I go with the Cummins, what information would I need to acquire in order to get genmon working there?

jgyates, 2021-07-02:
@derekatkins 

To add support for another generator the following things are needed:

1) a generator to test with, ideally more than one
2) The modbus definitions for the generator controller

In the past I have added generators that have a small handful of people interested (3 or more). I can leave this thread open to see if there is additional interest. If you have the register definitions you can send them as an attachment or link to this thread and I can look into how much effort this would take.

derekatkins, 2021-07-02:
@jgyates do you mean something like https://www.ccontrols.com/support/dp/modbus2300.pdf

jgyates, 2021-07-02:
yes, assuming this is the one for the RS40.

derekatkins, 2021-07-02:
The RS40 data sheet claims "PowerCommand 1.1", which is in that document.
I also found wiring data at https://powersuite.cummins.com/PS5/PS5Content/SiteContent/en/Binary_Asset/pdf/Commercial/Drawings/A041-J096.pdf

boeingpilot, 2021-07-23:
Not to hijack the thread, but have a similar situation.  In process of installing a Gillette liquid cooled 100kw.  It utilizes a Deep Sea Electronics 7420 controller.   I believe their Modbus Register definitions are here https://eccoengenharia.com.br/files/gencomm.pdf

I'm not a developer per se, but would be interested and where to start to adopt GenMon to support the Deep Sea Electronics series of controllers. Thoughts?

jgyates, 2021-07-30:
FYI, with version 1.17.0 I have added a feature that allows new controllers to be added with much less effort. Using this method you can get the notifications of generator state changes but more advanced features like remote start / stop and display of the internal controllers logs are not supported. More details are here:

https://github.com/jgyates/genmon/wiki/Appendix-N-Genmon-Supporting-Other-Controller-Types

derekatkins, 2021-07-30:
Cool!  Thanks.
It might be a good, oh, 9 months before I can test it!!

jgyates, 2021-07-30:
I am going to close this issue for now. Feel free to post here with additional questions or comments.

closed 2021-07-30 by jgyates

jgyates, 2022-02-03:
@boeingpilot 

FYI, the custom controller interface that was added has been tested on a Deep Sea generator: see this thread

https://github.com/jgyates/genmon/issues/580

and this wiki page

https://github.com/jgyates/genmon/wiki/Appendix-N-Genmon-Supporting-Other-Controller-Types

@derekatkins

If you are still interested in the Cummins support there are two example configurations that can be used as a guide to add support for you generator (assuming it supports modbus serial and you know the  modbus registers. If you need help let me know.

derekatkins, 2022-02-03:
@jgyates Thanks.  I am still waiting for the install.