jgyates/genmon #554: Two gauge enhancement problems

speters0, 2021-07-07:
Hi jgyates,
I'm sorry, I don't know how to contact you outside GitHub

Thank you for the integration!

I tried the last version for the two gauge enhancement.
Unfortunately the tweaked version did not work.

Here are the problems:
You can read the comments in the new modifications

In gaugediy.py 
function GetGaugeData 
the instruction time.sleep(1) is important you can read the comment in the new modification.
function PreReadCommand
CONFIG_VALUE_1 should be 0xD3 instead of 0xDC

In generac_evolution.py
function SetupTiles
after the comment: Wait for potential TankData to arrive 
the waiting code is important.
Because the TankData is coming from a separate Thread.
Without the wait, SetupTiles is executed before TankData is initialized

In gentankdiy.py
function TankCheckThread
there is a typo mistake.
Should be [tanktwo] instead of [twotaks]

I'm sending you the modifications.

Thank you!
[genmon.zip](https://github.com/jgyates/genmon/files/6774170/genmon.zip)

jgyates, 2021-07-07:
Thanks for the update. I see the need for everything you mentioned except the loop waiting on SetupTiles in generac_evolution.py. The reason this hsould not be needed is that:

Genmon.py will load and SetupTile will be called. Once gentankdiy.py is loaded it will call into genmon via the socket API which will trigger a call to /genmonlib/controller.py function SetExternalTankData which calls SetupTiles again if this is the first time calling SetExernalTankData. This should reinitialize the gauge data and include the external data. Also this avoids the up to a 10 second wait for instances that do not use external data. I checked in your update to gaugediy.py and gentankdiy.py. Let me know if this does not work and we can go from there. I would rather have SetupTiles be called again to initialize the gauges rather than the potential 10 second delay. This would make genmon.py not reliant on the add on settings (i.e. external gauge will work if the SetExternalTankData function is called via genmons  socket API, if the add on is not enabled it will function normally).

jgyates, 2021-07-07:
also, I did not update the version number with your last update, I just checked in gaugediy.py and gentankdiy.py.

speters0, 2021-07-09:
Hi,[genmon.zip](https://github.com/jgyates/genmon/files/6792723/genmon.zip)
I managed to find the time to look at it.

For an odd reason Python does not like the comments after

	if not tanktwo:
                CONFIG_VALUE_1 = 0xC3 # ...
            else:
                CONFIG_VALUE_1 = 0xD3 # ...

So I removed them in [gaugediy.py] function [PreReadCommand]

I also modified [generac_evolution.py] to include a second call to [SetupTiles]
in a waiting Thread initialized in [InitDevice] function.

You can simulate a two tank gauge by changing comments places in [gentankdiy.py] function [TankCheckThread]
and setting [nb_tanks = 2] in [/etc/genmon/gentankdiy.conf]

Now everything works fine.

You can look a the code, and tweak it if you like.[genmon.zip](https://github.com/jgyates/genmon/files/6792723/genmon.zip)
Thank you!

jgyates, 2021-07-09:
I made some changes to allow me to simulate the gauge stuff better without actually having a DIY gauge. I can now validate that two gauges show up (it does take a few seconds for the first reading to show up but it does for me). I removed the non ascii characters in the comments. Also, I validated the the latest change do allow for two gauges to show up if enabled in the conf file. I did not accept the change that adds the thread as this conflicts with other initialization code (SetupTile is called from other places so it is possible to have multiple threads waiting for several seconds to call Setup tiles again), also the thread is not explicitly killed on the cleanup/shutdown (not doing this leads to genmon not restarting due to the first copy not shutting down due to a thread not exiting when the app is killed) so I would rather just avoid this extra thread for maintenance and cleanup reasons if there is some other way to make the call to SetupTiles. Let me know if this works in your testing.

speters0, 2021-07-09:
You can notice that the Thread will end by itself in 100 seconds if it has not received TankData
Range can be as low as 6 seconds.

I'm receiving TankData Within 5 seconds on my Genmon dedicated RaspberryPi 3b

#-------------Evolution:WaitForTankData------------------------------------------
    def WaitForTankData(self):
        # ----- Wait for potential TankData to arrive           
        for i in range(100):
            if self.TankData != None:
                self.SetupTiles()
                break
            time.sleep(1)

speters0, 2021-07-09:
In Windows, child Threads are killed if the Main process ends.

I Don't know if it is the case for Unix.

jgyates, 2021-07-09:
In linux if you use kill pr pkill with the default options it will kill the main process but the threads should exit on their own. If you use the "kill -9" option it will kill all threads however it can take linux a few seconds to cleanup things like sockets so if you try to reuse them immediately you can get an error when using the "-9". what I have done to address this is to use a combination of the /genmonlib/mythread.py calls and the member function in /genmon/mysupport.py named WaitForExit. This allows you to use 1) wait until the thread exits on an event and 2) sleep in a thread for a given amount of time and exit the wait if the app has been killed.

If you wanted to add a thread in this code base, it would look like this:

     # this will start the thread with the MyThread class
    self.Threads["WaitForTankData"] = MyThread(self.WaitForTankData, Name = "WaitForTankData")

    #-------------Evolution:WaitForTankData------------------------------------------
     def WaitForTankData(self):
      # ----- Wait for potential TankData to arrive
     for i in range(100):
          if self.TankData != None:
                self.SetupTiles()
                break
          if self.WaitForExit("WaitForTankData",1):
                return

 In the Close method of the threads parent class if it uses WaitForExit instead of sleep, call this to kill the thread:

       self.KillThread("WaitForTankData")

The general idea is that each thread object in MyThread has and event that is joined to the thread that KillThread can wait on to make it synchronous and a Stop event object that is signaled when it is time to stop. WaitForExit will stop waiting if the stop event is signaled and return immediately.

The above code would be more in line with the way the code is doing things now however this would still not address the possible multiple instances of WaitForTankData being launched since SetupTiles can be called multiple times. One could move the launch of the thead to /genmonlib/controller.py in StartCommonThreads() and this would start this thread for all generator models, but since there was already a method to call SetupTiles again I just opted to use that.  There is a lot of history / evolution ( no pun intended) in the current code so I don't expect anyone to know all this stuff as it is very specific to this code base, but I thought I would share the thought process behind it.

I appreciate the contribution and testing. Thanks for helping with this.

jgyates, 2021-07-09:
I am going to close this thread for now but feel free to post here with any status or comments. Thanks

closed 2021-07-09 by jgyates