jgyates/genmon #555: Genmon not connecting

wrvolz, 2021-07-12:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

I'm getting the message : "Generator Monitor is not communicating with the controller at Generac" but when I bring up Genmon in the browser I'm seeing 13.00 volts for the Battery and 249 volts for the utility. If it weren't communicating would I still be seing this? This occurred when I updated to the latest version and was working fine before that. I sent logs to the developer. It sometimes connects and then stops comms again about 2 minutes later. Is there an issue here? When I was at the Genmon page at the Monitor tab and then clicked back to the Status tab, the utility voltage had dropped to 246 so it's getting something.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

Maintenance :
    Model : G0062500
    Generator Serial Number : 0008370131
    Controller Detected : Evolution, Air Cooled
    Nominal RPM : 3600
    Rated kW : 20 kW
    Nominal Frequency : 60 Hz
    Fuel Type : Propane
    Estimated Fuel In Tank : 764.04 gal
    Hours of Fuel Remaining (Estimated 0.50 Load ) : 322.38 h
    Engine Displacement : 999 cc
    kW Hours in last 30 days : 0.03 kWh
    Fuel Consumption in last 30 days : 0.60 gal
    Total Power Log Fuel Consumption : 35.96 gal
    Run Hours in last 30 days : 0.50 h
    Fuel Level State : OK

    Controller Settings :
        Calibrate Current 1 : 1470
        Calibrate Current 2 : 1453
        Calibrate Volts : 1010
        Nominal Line Voltage : 240 V
        Rated Max Power : 20 kW

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2021-07-12:
Your logs indicate you are having serial connectivity issues. I would start with the troubleshooting steps specific to serial communications listed here:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

Communication issues can be intermittent, for example a marginal connection or short can make the communication issues come and go depending on temperature or vibrations. The Monitor page should list the types of communication errors you are seeing.

wrvolz, 2021-07-12:
I jiggled the connection at the generator - seems that's a bit loose and/or got some moisture in it. You can close the ticket I opened on this. Thanks,

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting [https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting]

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

closed 2021-07-12 by jgyates