jgyates/genmon #556: This is a great project (thanks)!  Looking for contributors to help port to an exiting ESP32 board (Arduino IDE Compatible) with onboard Wifi and 4G support...

evanevery, 2021-07-15:
This is just an FYI to see if anyone is interested in working on a project to implement similar technology using a custom PCB with an ESP32 (w/Wifi and Bluetooth) and an integrated SIMCOM 7600A (4G SMS/DATA and GPS) support.  **Full disclosure**, this EXISTING board is one that I personally designed and manufacture:

[https://www.haute-solutions.com/prod_pcb.html#ESP324G](url)

I have been looking to replace the proprietary 3G (and now WiFi) Generac monitoring solutions with something Open Source for several years now.  I have the 3G unit (no longer supported) and was waiting for a WiFi enabled replacement but now am told they are back ordered for several months (due to the "chip shortage").  So this prompted me to go back out and see if any additional info has become available on the wiring pin outs for the 8-Pin accessory port.  I had pretty much identified the two power pins and the two data pins (from online photos and manuals) and assumed it was using RS-232 but had no idea of the baud rate or messaging protocols.  Then I found this project.  Very Nice!  Thanks!

While I think the Pi is a very capable platform for this project, I would like to use one of my ESP324G boards as it can add the ability to do bi-directional SMS communication (Notification, Reporting, and Control) via the integrated cellular capabilities.  The ESP324G also supports power via 12V natively and is fully programmable via the Arduino IDE.  I have several fully developed (open source) libraries for high level abstraction of this boards SMS capabilities which also add support for an OLED display (follow the product link above).  I think the ESP324G would make a very nice single board solution which adds SMS support to the package.  Effectively, the ESP324G could be mounted remotely on the Generator (just like the 4G or WiFi options) except that it could provide BOTH levels of functionality (WiFi and 4G) and be completely independent of any proprietary service or subscription requirements!

Although I am not a Python programmer (I prefer C), I expect it shouldn't be too hard to parse the info I need from the code in this project.  If anyone would like to join me in a spinoff Open Source project using the ESP324G, please reply to this thread so we can work together.  (I am happy to send a free ESP324G board to anyone who is seriously interested in contributing)

Thanks!

jgyates, 2021-07-15:
I would recommend looking at this wiki page if you are going to port to a different platform:

     https://github.com/jgyates/genmon/wiki/Appendix-E---Running-on-Non-Linux-Platforms

If you are not running linux then using a TCP/IP stack that is compatible with python or POSIX would be helpful. If you just want the serial functionality then you could strip out a lot of the other features.

evanevery, 2021-07-15:
Yeah, I've been looking through the code.  Very full-featured!

I'm very familiar with the TCP stack available via the Arduino IDE libraries.  Also with the WiFi application as well.  I've written many TCP (Ethernet and WiFi) aware apps for devices via the Arduino IDE.  I expect that to be a fairly trivial part of this exercise.

The aspects I would be least familiar with would be the modbus protocol and its application for this device.  However, you seem to have everything very well documented so I just need to work to understand how to extract the details from Python Source.  I've written (to one degree or another) in dozens of languages but never needed to use Python (yet).  Not worried about it but I'll have to figure it out along the way.  I'm not looking to replicate this project, just extract enough of the groundwork you have already done so I can get a rudimentary foundation up and running in the Arduino IDE (using C).  

All the back end communications (TCP/WiFi/SMS/etc) is relatively trivial given the code and libraries I already have in place.  Its the communications protocols unique to the device which I need to extract.  I'm sure I can find everything I need in your code base!

As is my style, I'll start off small to get a handle on things.  Build the power/com cable and interface to get the power and 3.3v RS232 sorted up to the ESP324G board and then start "poking at stuff"...  My immediate goal is to simply get notified (via SMS) when the generator turns on/off and why (exercise or power failure).

Once I get my board talking to the generator the rest should be all downhill...

granttoth, 2021-08-02:
You may be interested in [some code I posted a while back. ](https://github.com/jgyates/genmon/issues/399)  The code has changed a bit since then.  I am currently using defines for the register addresses and I've combined multiple conditions for my alarms.  As an example; I have all the oil alarms triggering the same relay.    My next iteration of my project is planned with an ESP32 and I can't wait to see what you come up with!

jgyates, 2021-09-02:
I am closing this thread since it is not active but feel free to post here if there are any updates to this topic.

closed 2021-09-02 by jgyates