jgyates/genmon #557: Monitor: Monitor Health: Not receiving data

bcrazycramer, 2021-07-16:
### Expected Behavior

Monitor Health: OK

### Actual Behavior

Monitor Health : Not receiving data

Timeout Errors: 5844

I replaced the RS232 to TTL connector. We have had some of these fail but on this generator I can still go to Status and see the battery and utility voltage. 

- Generator Model: G0067292
- Genmon Logs: Submit via the About page
- Genmon Version: V1.16.13

bcrazycramer, 2021-07-16:
I submitted the logs via the About page.

jgyates, 2021-07-17:
Serial connections can be intermittent. I would look at this issue first as it has intermittent  serial issues as well

     https://github.com/jgyates/genmon/issues/555

Your logs indicate you have serial timeouts which, if intermittent would cause some data to be displayed properly but enough of these errors would cause the monitor health to show the status you are seeing. Check your cabling, including the ground connection from the pi to the generator controller.

jgyates, 2021-07-19:
any update on this? Have you tried the serial troubleshooting steps?

bcrazycramer, 2021-07-19:
One of console wires going into the plug for the generator was loose. It's been working for about 2 hours now without any issues.

jgyates, 2021-07-19:
I am glad it is working now.

closed 2021-07-19 by jgyates