jgyates/genmon #558: Not receiving Data

kmessin357, 2021-07-19:
Hi I have I have replaced the RS 232 converter board as it did appear as it was working. Have done troubleshooting for serial port and it says it ok. I have obviously missed something. I have dumped and reloaded Genmon multiple times thinking something didn't load right, or I missed something.

I have firmware V1.11 and Hardware V1.00 

### Expected Behavior

Monitor Health : System Initializing. 

### Actual Behavior

Not receiving data.

### Your Environment

- Generator Model: 
- G0070381
- Genmon Logs: Logs are empty did submit registers
- Genmon Version: V1.16.13

jgyates, 2021-07-19:
Unfortunately I don't have nearly enough information to make any recommendations. I have not received your logs or registers. Based on your description you have done some troubleshooting but I don't know what you have tested but apparently it passed.

Have your went through these steps?

     https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

Which test pass and do any fail? Each test validates a separate part of the serial communications.

If your outbound email is not working then the logs will not be submitted for review.  If you have outbound email setup then load genmon, wait 5 minutes then submit your logs.

If you logs are completely empty then you are either looking at the wrong logs or the software is not loading. The logs have info that can provide insight into what is or is not happening. It also describes the type of serial failures if any.

kmessin357, 2021-07-19:
Sorry,
I did 3.6 serial-troubleshooting no errors all passed. I have shutdown and
restarted, waited and resent logs. I do however have a low voltage warning.
I resent registers and logs. I have attached a picture, maybe that will
help. This has not been running. I only get info as you see on attached
photo no data from generator.

Thanks

skipfire, 2021-07-19:
@kmessin357 how have you grounded your setup?  Have you tried flipping the RX & TX between the converter and the generator?  I had them backwards when I first installed but the loopback would still pass in that case.

jgyates, 2021-07-19:
@kmessin357 

Please describe the details of your setup. Your logs show CRC errors. Are you using serial or serial over TCP (you have both logs so you at minimum have enabled serial over TCP at some point). If you are getting CRC errors you either have serial over TCP configured incorrectly, or you have not installed the proper libraries.

Your low voltage and throttling errors are also concerning. How are you powering and grounding the raspberry pi. THis could cause a variety of errors.

Also, you cannot attached pictures when responding in email to this thread. If you do they will be removed by github. To attach pictures you must use the github web interface and drag the picture to the comment pane.

jgyates, 2021-07-19:
From you log submission:

Comm Stats : 
   Packet Count : M: 370, S: 0
   CRC Errors : 330 
   CRC Percent Errors : 89.19%
   Timeout Errors : 40
   Timeout Percent Errors : 10.81%
   Modbus Exceptions : 0
   Validation Errors : 0
   Sync Errors : 0
   Invalid Data : 0
   Discarded Bytes : 0
   Comm Restarts : 0
   Packets Per Second : 0.80

This indicates you sent 370 packets (so far) and received zero valid packets. 
330 CRC errors which indicates you received data but the modbus CRC is incorrect. CRC errors are treated as invalid packets
40 timeout errors (sent a packet but did not receive anything)

kmessin357, 2021-07-19:
I have powered it from the battery in the generator. I have tested through
the wiring from pi to converter and it tested fine. I have reversed the
wires and that made no difference. This generator came with wifi so I'm
guessing the original plug from wifi is wired correctly. It has 4 wires,
blue white red and black. The blue and white wires are where i connected to
the genset controller. I would think the red is positive power and black is
negative. Should I have connected ground to that black wire instead of
using the battery power?

kmessin357, 2021-07-19:
> If you connected to where the wifi wires were you are connected to the
> wrong spot as the wifi runs on RS485 pins not RS232 pins which are the
> other two on the "top". Bottom two are power, but the 5v under the RS485
> pins I believe does lead to the undervoltage issues, and under the RS232
> pins is 12v & GND.

The original generac WiFi used a moldex 8 pin connector, and was plugged
into the underside of the Evo 2 panel just below the digital readout
display . It used pins 3,4,7 and 8. I connected the TX to Pin 8 and the RX
to pin 7. I would think the other two black and red on pins 3 and 4 would
be power. I did not use the black and red wire pins. When looking at the
diagrams it looks like I have the right place. There were no wires
connected to pins 1,2,5, and 6. Based on of this information does it
sound like I am connected correctly?

Thanks

jgyates, 2021-07-19:
How are you powering the pi? You mentioned using the battery but that is not enough detail to determine if your power is setup correctly. Are you using the USB 12V converter or something else? Something is not right with your setup so these details are needed to figure out what is missing. Pictures of your setup posted to this thread may help also.

skipfire, 2021-07-19:
You should not use any of the wifi wires, except maybe ground.

kmessin357, 2021-07-19:
I am using PI 3B+, powering with Motopower MP0609AA 3.1 Amp USP powered
from the generator battery.

Using a Uxcell RS 232 D sub db9 female adaptor to terminal connector, DZS
Elec RS 232 db9 male serial port to TTL converter Max3232, and startech eps
8 pin power extension cable using pins 7 and 8.

I am using your instructions for setting up serial port. There is a
possibility that something could have been configured wrong. I am not
proficient in python for sure. Sorry, I wasn't aware that pictures were not
allowed.

Thanks,

kmessin357, 2021-07-19:
Sorry, I didn't connect to the wires. I cut the wires from the original
wifi moldex plug that was plugged into the EVO 2 underside, and connected
the wires from terminal 2 and 3 from the converter to pins 7 and 8 on the
original moldex plug and then plugged that back in to the original spot.

jgyates, 2021-07-19:
Does this mean that it is working for you now?

skipfire, 2021-07-19:
The pin numbering in the instructions is the evo2 connector or the back of the male molex connector (where the wires come out), The WiFi module does not use the RS232 serial pins that genmon uses, so if you just cut the wifi cable and used the pins in the molex without moving the pins, then you are connected to the wrong pins.

skipfire, 2021-07-19:
Also, you can use pictures, but they must be uploaded through the web interface.  Attached is my working cable for you to reference.  The yellow wires are data pins. 
[image]

kmessin357, 2021-07-19:
Thank you for the picture. Those are the two pins that I am using.

kmessin357, 2021-07-19:
No, I think we are now on the same page for the pins that are used on evo2.
they are on the right pins.

skipfire, 2021-07-19:
Verifying ground is common is the only other thing I can suggest.

kmessin357, 2021-07-19:
Question, If I am powering from the battery positive and negative into the
Motopower that is converted to usb for the Raspberry Pi, and using pin 06
on the Pi to the ground of the serial port converter, that should be
considered a sufficient ground would it, or do I need to run another ground
wire somewhere else? Everything other required ground should be running
back to the battery, including the evo2 controller right?

skipfire, 2021-07-19:
Check a multimeter to make sure that ground on the Pi to ground on the battery is 0, or continuity check the same.  If you check voltage to positive you will likely see 12.8 to 13.4 due to the charger voltage.  You could wire the ground pin into ground on the serial adapter.

jgyates, 2021-07-20:
I would recommend an additional ground. While I don't know what is inside the USB 12V to 5V converter it could be a DC to DC converter and use a separate ground on each side (12 vs 5 volt side)

kmessin357, 2021-07-20:
Ok, thanks, I will do that tomorrow. Its raining here now and something
about water and electricity I don't like, especially if they get mixed.

kmessin357, 2021-07-20:
Ok, thanks, I will do that tomorrow. Its raining here now and something
about water and electricity I don't like, especially if they get mixed.  I
think you already received this, but I'm just making sure. I appreciate
what you two are doing for me to get this resolved!

kmessin357, 2021-07-20:
I checked the ground and it checked out ok. I  added a ground wire also,
and there is no change. Is it possible it is encrypted.  I had the original
wifi disconnect in March on 2021.

jgyates, 2021-07-20:
Your logs are showing that when a timeout occurs the value transmitted is in the receive buffer. This indicates that you could have a short between your transmit and receive pins (or you left the loopback wire connected). Can you measure the continuity between the tx and rx pins on your pi. Check them at both the RS-232 side and the raspberry pi side. This could examplan the odd behavior you are seeing. Based on your logs, this is my primary theory.

Even if you have a controller the encrypts the data, you should not be seeing the results you are seeing.

Also, have you modified the config files by hand in any way? If you have I would recommend replacing them with a fresh copy as this may lead to unexpected behavior if you have an invalid entry in your config file.

kmessin357, 2021-07-20:
I did continuity check on pi pins 8 and 10 they are open, so are tx and rx
on 232. Could you please send me the commands that are needed to replace
the config.file. That way you know that is correct. Will I have to make any
changes to it or leave as is?

jgyates, 2021-07-20:
ssh into the pi then:

      cd genmon
      sudo cp ./conf/genmon.conf /etc/genmon
      ./startgenmon.sh restart

to exit ssh type:

     exit

kmessin357, 2021-07-20:
Ok, that is done, but still no change. The PI  under voltage is flashing
intermittently now, I had not noticed that before, but may have been doing
it. CPU Frequency Throttling is intermittently active also.

jgyates, 2021-07-20:
I would check the voltages at your battery and the 5V output of the converter. You should have a bad voltage converter. If the CPU frequency throttling is bad then that could cause an extreme slow down of the CPU which could cause issues with the serial timing

kmessin357, 2021-07-20:
I checked that when I was checking the ground. 13.6 V on battery and 5.06 V
out of converter.

kmessin357, 2021-07-20:
Is it normal for the red power LED on the Pi to flash on and off when
powered up?

jgyates, 2021-07-20:
No, that is not normal. The red LED is the power indicator. The green LED is the SD card activity. You could have a faulty power cable.

also, you can try running the loopback test without a loopback installed:

disconnect the cable from the generator, then ssh into the pi:

        cd genmon
         ./startgenmon.sh stop
         cd OtherApps
         sudo python serialtest.py

post the result of the test here

Then connect a loopback to the end of the cabled, connecting the tx to the rx pins, then type

       sudo python serialtest.py

post the results here

to restart genmon reconnect the cable to the controller then type:

          cd ..
          ./startgenmon.sh restart

kmessin357, 2021-07-21:
All answers in order in which you requested.

testing for serial port /dev/serial0...

Serial port opened
write date: sent test string
waiting to received data....
Failed: Sent data does not match receive. Received 0 bytes

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data...
PASSED! Loopback successful

Restarting genmon python scripts
Stopping....
Starting....
Starting /home/pi/genmon/genmon.py
Starting /home/pi/genmon/genserv.py
***@***.*** : ~/genmon $ Serial port opened
-bash: Serial: command not found
***@***.*** : ~/genmon $ write data: sent test string
usage: write user [tty]
***@***.*** : ~/genmon $ waiting to received data....
-bash: waiting: command not found
***@***.*** : ~/genmon $ FAILED: Sent data does not match receive.
Received 0 bytes
-bash: FAILED: command not found
***@***.*** : ~/genmon $

jgyates, 2021-07-21:
that looks as expected.

if your red LED on the Pi is blinking or going on and off they I would look your power cabling. The red LED should remain solid.

kmessin357, 2021-07-21:
I will try another cable in the morning.  Thanks.

kmessin357, 2021-07-21:
Morning, I replaced the power cable and now have a steady power indicator
on.

kmessin357, 2021-07-21:
New cable fixed the intermittent power LED, but still have com problem.

jgyates, 2021-07-21:
try reinstalling all of the libraries:

ssh into the pi:

        cd genmon
        ./startgenmon.sh stop
        ./genmonmaint.sh -i

Type Y to begin the install, you can answer Y (yes) to every question except "start genmon on boot" since you likely already have it set to start on boot and you don't' want to start it twice. If you select Y for copy conf files then it will overwrite all of you config files.

Once this is done you can reboot the pi:

       sudo reboot now

kmessin357, 2021-07-21:
I tried to reinstall it got all the way through the process right to the
part where I did the sudo reboot now and putty had a fatal error. I got
back out and got back into putty run through the whole install process
again and I got the same fatal error. Now I can't get back in. I tried 3
times and nothing happens. I went to the generator to power the Pi down
waited a couple minutes and powered it back up. Tried again to putty in no
luck. I tried to get in with VNC and can't get in that way either. Just
times out from the website using IP address. It may have corrupt the sd
card or program.

jgyates, 2021-07-21:
was your SD card new and a class 10 or better SD  card? If not then you problems could be related to the SD card.

kmessin357, 2021-07-21:
Yes, It is a SanDisk XC 64GB Class 10.

jgyates, 2021-07-21:
it is possible that your power problems have corrupted the card. A reformat and reinstall of the operating system may be in order.

kmessin357, 2021-07-21:
I went out and powered off again it came so I run the install again. When I
do the Sudo reboot I get a message puTTY Fatal Error that says
X Remote side unexpectedly closed network connection.

jgyates, 2021-07-21:
The sudo reboot will reboot the pi so the pi closes the ssh connection to puTTY so that is normal if this happens after the reboot command.

kmessin357, 2021-07-22:
I'm  reinstalling everything again on a different  SD card.

kmessin357, 2021-07-22:
Remove SD card from Pi and replaced with another new one. Formated, and
reloaded all software.  Just reinstalled Pi in the generator and
started, sent you logs. I was hoping this was going to do it. Still not
receiving data.

skipfire, 2021-07-22:
@kmessin357 any chance you can share a photo of your cable?  You will have to do that through the github website rather than an email reply though.  Just wanting to rule out anything due to starting with the WiFi molex.

jgyates, 2021-07-22:
as a test I loaded genmon on a pi only connecting Tx to Rx and I got the exact results you are seeing (mostly CRC errors with a few timeout errors). Either something is connected wrong in your cable, your serial hardware is bad or your controller is not functioning as expected on the serial interface.

Try running the loopback test while connected to the controller. This *should* fail, but I am guessing it will pass.

jgyates, 2021-07-22:
repeat the test above but do not disconnect from the controller and do not add a loopback. You still need to unload genmon to do the test.

jgyates, 2021-07-22:
I agree with @skipfire, pics of you cable would go a long way

kmessin357, 2021-07-23:
I am taking a second look at your picture and it looks like my wires are on
the opposite side. l need to pull the pins and reinstall. I will send an
update after I do that.

kmessin357, 2021-07-23:
I took a second look at the molex plug wiring and I was trying to go off
memory, and you were correct the original plug for wifi IS DIFFERENT  then
the required wiring for this project. I rewired and put back in and it is
working!! You two are wizards! Thank you for being so supportive and to
keep pushing me back to that plug, Awesome!! I'll send logs.

skipfire, 2021-07-23:
@kmessin357 I'm thrilled it is working.  Any chance you could send me an email?  I'm going to be selling a board (hopefully next month) that would have made this a 5-second plug & play setup and I'd like to get your feedback on it.
helmsjr at skipfire.com

kmessin357, 2021-07-23:
No problem, where would you like me to send it?

skipfire, 2021-07-23:
@kmessin357 helmsjr at skipfire.com

kmessin357, 2021-07-23:
[image]
Here is a picture of the Original wiring of the com plug. You were right working fine now. Thanks for all your help. Issue can be closed.

closed 2021-07-23 by jgyates

kmessin357, 2021-07-23:
Thanks again for all your support. I appreciate it, you guys are awesome!
And looks to be a fabulous program!