jgyates/genmon #560: Current Transformers and Run Hours

philruel, 2021-07-22:
Hi,
I have a Generac 22 KW air cooled generator and it doesn't have Current Transformers.  I would like to purchase some from Generac but looking at parts diagram for smaller generator, I don't ever see them listed.  Can someone show me the parts diagram with CT1 and CT2 with their part numbers.

Also, my generator has 322.5h and genmon shows: Total Run Hours : 322.00 h.  Is this normal?

Thanks you

BTW, genmon is amazing!! 

- Generator Model: {G0065520}
- Genmon Logs: Submit via the About page
- Genmon Version: {1.16.13}

liltux, 2021-07-22:
@philruel 
CT's for EVO1 20KW - 0K0508
CT's for EVO2 24KW -  Same number comes up, 0K0508

jgyates, 2021-07-22:
Yes, for and Evolution air cooled the tenths are not reported via modbus. What you are seeing is expected.

philruel, 2021-07-23:
Thank you for your answers.

@liltux 
Are you aware if the Current Transformers come with wire and terminal ready to be plugged in the main controller?

Also, I removed the fuse from the control panel and the battery stopped charging and the voltage started dropping, as expected.  The voltage went down to 13.2V and still dropping but Genmon continued to report a voltage of 13.9 - 14.0V.  How come?

Thanks again!

liltux, 2021-07-23:
@philruel yes wires are built on the transformers, the pins I would assume but have never ordered one to know for sure.  The controller sees both battery and charger voltage, when the battery is charging the charger is connected to the battery via internal relay contact.  It sounds like the register being monitored is tied to the battery charger voltage rather the actual battery voltage.  May be useful to do a mod is dump while removing and installing the fuse to try and find the correct battery voltage register.

@jgyates your thoughts on this?

jgyates, 2021-07-23:
@liltux  knows much more about the internals of the generator that I do so I would defer to him on this.

grsthegreat, 2021-07-27:
i simply ordered ct's from amazon. they came with wires, but not the pins. when you price out the 0k50508, you'll fall out of your chair.
https://www.amazon.com/gp/product/B01LWN37KS/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

the one i got shows currently unavailable, but any 100 amp CT will work.

philruel, 2021-07-27:
Would the terminals be the same as the wire highly recommended for Genmon? (https://www.amazon.com/gp/product/B000M802RG)

If so, I would have to remove 4 terminals, splice them to the current transformers then enter the terminals in the J1, J2 connector.
So no Molex crimping tool needed?

grsthegreat, 2021-07-27:
I have no idea whet you are referring to. The ct’s just wrap around the 2 wires going to the breaker in the generator then plug into the controller. 

 

The location where they plug in depends on whether you have an evolution 1 OR evolution 2 controller. They plug into different ports.

 

I simply added some stranded #14 wire to the wires included on the ct’s and then soldered a molex connector to that. 

 

There was nothing needed to remove with a molex removal tool, though I do have one.

 

 

Go to this web site. Theres detailed directions 

https://github.com/jgyates/genmon/issues/403

grsthegreat, 2021-07-27:
https://github.com/jgyates/genmon/issues/403

please read this to fully understand procedures

philruel, 2021-07-29:
Thank you for the clarification.

What I was referring to is the section 3.1 Making a Cable for Genmon.  jgyates highly recommended buying this (https://www.amazon.com/gp/product/B000M802RG) that plugs directly in the controller and not having to deal with buying terminals and a crimper.

So, what I was thinking is buying that cable again, cutting the end that is not good and removing the terminals from the Molex connector on the good side (male or female).  Then, I'll have 8" of wire with a crimped terminal that will be spliced to the CT.

What I don't know is if J1, J2 and the connector we use for Genmon have all the same terminals?

Thank you!

grsthegreat, 2021-07-29:
Oh, i see. The thing is i have no idea if they have the same terminals. Maybe someone else here does.

jgyates, 2021-07-31:
I don't know either.

jgyates, 2021-08-04:
I am going to close this thread from now but feel free to post here if you have addition questions or info on this topic.

closed 2021-08-04 by jgyates

philruel, 2021-09-09:
I purchased from Amazon 2 Current Transformer PZCT-2, same as people on the board suggested.

I tested them on my clothes dryer circuit.  The dryer specs 230 V, 24 Amps.  Results with my clamp meter, average of 23.6 V.  All is good here.

I hooked my PZCT-2 to the wire of my dryer and set my multimeter to VAC and got an average of 15.6 VAC
In the repair manual p.80, 25 Amps = 0.380 mVAC.  I’m way over this value.

What am I not understanding or doing correctly?  I don't want to fry my controller.

jgyates, 2021-09-09:
Are you referring to your dryer repair manual? The CTs should output a current that is a defined fraction of the current used on the line monitored. The voltage would vary based on the resistance of the circuit it is attached to. Typically CT would need a specific range of resistance to get a meaningful value. Based on what you are describing I don't have a good picture of how things are connected. Do you have any resistor between the CT leads? The spec on the CT should give the recommended value.

philruel, 2021-09-09:
Sorry, no I'm referring to the Generac repair manual p.80 which says at 25 Amps = 0.380 mVAC for the CT.

To test the CT, I used my dryer which pulls 24 Amps

I connected the red wire of the CT to the red wire of the multimeter.  Same with black wire.  
Set the multimeter to VAC and took my reading.  I don't have a resistor between the CT leads.

I was expecting a voltage of ~0.380 mVAC since other members used this CT.  But my voltage is 15.6 VAC.

Thank you

jgyates, 2021-09-14:
Unfortunately I don't have any first hand experience with these specific CTs. I have used CTs on an open energy monitor based project but my generator is not an air cooled model so I don't have a direct reference point.