jgyates/genmon #561: Cable documentation update

skipfire, 2021-07-23:
I'd like to suggest adding a bit of text to the cable documentation about the WiFi plug.

For units equipped with a Generac WiFi unit, it is not recommended that this plug be repurposed for genmon.  The WiFi uses 4 pins, none of which are the RS232 data pins needed for communication with the Raspberry Pi.  Using that plug would require re-pinning the molex connector and since doing so would require cutting the WiFi conductors you would be doing irreversible modifications to your generator.

liltux, 2021-07-23:
@skipfire, I am not clear on which wires you would have to disconnect? 

> Using that plug would require re-pinning the molex connector and since doing so would require cutting the WiFi conductors you would be doing irreversible modifications to your generator. 

The pin ports 7 and 8 are used for RS232 and are not populated?  Also, I would not advise someone to use the power port pins for accessory power on the EVO2, if that is what you mean to disconnect.  I have found through testing with a pi0, if the EVO2 does not communicate to the WiFi Module I have seen it turn that accessory port power off.  At least it did with my pi zero during testing.  All one would have to do is add the two wires and pins to the pin port 7 and 8 for RS232.  In essence an end user could do both. 

One could use the same pins for RS485 as the WiFi module though.  Same serial data just sent through a different protocol.  @jgyates I believe was quite successful at getting data from those 2 pins if I recall.

I agree an update maybe useful but it would need to be clear and correct information.  See below for correct pinout of that connector.
[image]

[image]

skipfire, 2021-07-23:
@liltux the WiFi module only has 4 conductors (pins 3,4,5,6) so if you want to use it with genmon you need to repin 5 & 6 to 7 & 8 at a minimum, also potentially a ground.  I've been using the 12v for a few months for power with my prototype boards on an EVO2 controller and no problems.

liltux, 2021-07-23:
@skipfire I don't think you are following me.  All one has to do is add additional wires.  Those molex pins are available on Digikey for about $.16 each.  You only need 2 to add a wired connection to 7 & 8.  I did not try to power from 12V source as I was not using an onboard power step down.  I was successful at using the 5 VDC on the EVO1.  If using 12V I would prefer going straight to battery, preference.  An extra ground is not needed.  In all reality, if you have a good connection to common ground that is not the issue, its common.  

I am not saying you can not do it your way, just the verbiage should indicate all options that are viable.

skipfire, 2021-07-23:
The key piece of information I was suggesting is that the 4 wires on the WiFi are not in the spot they need to be for the rest of the instructions, a recent setup problem came down to that being the problem.  I have nothing against listing all the potential solutions.

liltux, 2021-07-23:
Someone out there is getting a kick out of this thread.  I just read the WiKi, it details all the correct information.

skipfire, 2021-07-23:
I agree that it has correct information and nothing wrong.  I'm suggesting an explicit "if you are trying to reuse the WiFi molex" note from recent experience on #558 .

jgyates, 2021-07-23:
I added this to the wiki page on making a cable:

If you have an Evolution 2.0 unit with the wifi module you have a few choices. 1) Do not use the wifi module (unplug it) and directly connect the pi to the molex connector. This option will not allow your firmware to automatically updated via wifi. This may or may not be desired depending on your perspective. Updating your firmware in the future could prevent genmon from working if future versions of the firmware include features designed to prohibit gnemon from working, or 2) Use the wifi module with genmon by re-pinning the connector  since the wifi module uses separate pins on the molex connector than the pi uses. Care should be taken if you attempt to modify the existing molex connector on the wifi module as any mistakes could damage your connector.

Let me know if you guys thing this needs anything else. Thanks for the feedback

closed 2021-07-23 by skipfire