jgyates/genmon #562: Forgot password

DJT1423, 2021-07-25:
I have recently changed my monitor password and now it appears I have forgotten it. Is there a password reset function, or can you advise where the password settings are located in the file structure?

BTW, I love the project. Thanks so much for developing and maintaining this.

Thanks, 
David T

jgyates, 2021-07-25:
if you ssh into the pi you can edit the /etc/genmon/genmon.conf file with root privileges:

     sudo nano /etc/genmon/genmon.conf

The entry you are looking for is 'http_user' and 'http_pass':

        https://github.com/jgyates/genmon/blob/master/conf/genmon.conf#L204

 Make any changes, save the file and then restart genmon.

       cd genmon
        ./startgenmon.sh restart

Then login from the web interface with the new password.

DJT1423, 2021-07-25:
Great, thanks. Good to have this info filed away.

David

David Thompson

***@***.***

    https://github.com/jgyates/genmon/blob/master/conf/genmon.conf#L204

closed 2021-07-25 by jgyates