jgyates/genmon #563: Calibrating power output

DJT1423, 2021-07-26:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected behaviour

{Please write here} Expecting power output readings greater than I'm seeing

### Actual Behavior

{Please write here} Power output never exceeds 1KW even when air conditioner is running. Seems inappropriate. 

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here} 
[image]

### Your Environment

- Generator Model: Generac 11KW
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here} v1.16.13

jgyates, 2021-07-26:
This page list known issues, specifically number 5 describes how to modify the settings to account for CTs that have not been properly calibrated.

     https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

DJT1423, 2021-07-26:
Yes, I read this:

"If the default values used are not working for your generator please let us know. You can override the value used to divide the register value, on Evolution Air Cooled Models, with the 'currentdivider' parameter in genmon.conf."

but I can't say I understand what I am supposed to be doing. Is it possible to provide more explicit instructions?

David T

 https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

jgyates, 2021-07-26:
You can view the modbus registers on the registers status page. Single click the gear icon on the top right on the web interface to view the register status page. If you double click the gear icon it takes to to the advanced settings page. The register status pages shows the contents of the modbus registers pulled from the controller. On the register status page you should see register 003a. This value is used to calculate the current, which then combined with the voltage reading to determine the power being consumed by the generator. There are two other settings that are also used: current offset and current divider. They are used like this:

     Current Output = (Modbus Register 003a Values + currentoffset) / currentdivider

The values for current offset and current divider are fixed for each model.

The first thing to try is to go to the advanced settings page and enable "Use Legacy Power Calculation", save the value at the bottom of the page, the test to see if this change makes a difference in the results you are seeing.

If this does not give the expected results, then you can probe further, but it will require some measurements from a clip on multi-meter (to measure current). If you can measure the current in each leg of your house power when the generator is running and your transfer switch is active then add up both legs. While keeping the load on the circuit consistent as possible, look at the register page and make a note of the value of registers 003a. Use the formula above to determine the needed values of currentdivider and currentoffset (if needed) and plug in these values to the currentdivider and currentoffset values on the advanced settings page. Then see test again to see if the current measured under load with the meter is close to the value reported by genmon.

If you are not comfortable making measurements on high voltage lines you can disable the power reporting on the advanced settings page or just leave it as it is.

Let me know if you have any questions.

jgyates, 2021-07-26:
Another option, if you have the maintenance manual for your generator it may contain information on how to calibrate the current transformers in your generator. There are two values that you can change in your controller (calibrate  current 1 and calibrate current 2) that adjust the readings of the two internal current transformers. I am not familiar with how to do this but there is a procedure in he maintenance / repair manual. That would possibly address the problem from the other side (controller side vs genmon side) of things.

grsthegreat, 2021-07-27:
to calibrate current transformers in the evolution 1 or 2, you need to access the dealers menu. 

if someone screws up a setting in this menu you can theoretically wreck your controller or the generator itself. personally though i service 100's of units, i have never re calibrated a ct.

grsthegreat, 2021-07-27:
go to this post and you will see detailed descriptions and some photos.

https://github.com/jgyates/genmon/issues/403

jgyates, 2021-08-04:
I am going to close this thread but feel free to post here if you have questions or comments on this topic.

closed 2021-08-04 by jgyates