jgyates/genmon #564: Another "System Initializing. Not receiving data" problem (sorry!)

eBoon123, 2021-07-30:
### Expected Behavior

New install of Generac 18KW generator with Evolution 2 controller - expecting to see data from the generator

### Actual Behavior

Monitor page shows: Not receiving data

### Steps to Reproduce (including precondition)

Look at web page

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac Guardian 18kW G007226-0,
- Genmon Logs: Submit via the About page
- Genmon Version: V1.16.13

First off - thank you for a great project - very excited to get past this issue and explore all the data on my new generator!

Just had my generator installed yesterday and have set up Genmon on a Raspberry Pi 3. I have reviewed the troubleshooting guide and I am not having any luck getting data to genmon.

I am powering the Raspberry Pi off of the battery using a MotoPower 12V to USB module.

Here are the results of serialconfig:

[image]

Here are the results of serialtest - the first test without the loopback in place and the second (successful!) with the loopback in place :

[image]

Here are pictures of my cable and connections inside the box:
[image]
[image]
The black and red wires are spliced into the yellow wires in the above picture. Black is going to Pin 8 and Red is going to Pin 7 on the Molex connector. I also switched the Red and Black wires around - same results.
[image]
[image]

I have also just sent logs via the About Page.

Thank you in advance for any help!!

skipfire, 2021-07-30:
The pic doesn't show how the yellows go to the red/black on the screw terminals , but TX and RX may be backwards, loopback will work but it will fail after.

eBoon123, 2021-07-30:
> The pic doesn't show how the yellows go to the red/black on the screw terminals , but TX and RX may be backwards, loopback will work but it will fail after.
Black is going to Pin 8 and Red is going to Pin 7 on the Molex connector. I also did switch them around just in case...unfortunately did not help.

Thanks for the reply!

skipfire, 2021-07-30:
You might also try connecting ground on the molex.

jgyates, 2021-07-30:
I agree with @skipfire, connect pin 2 on the molex (Ground) to pin 5 on the DB9 modular connector. This will connect ground from the controller to the RS-232 board.

eBoon123, 2021-07-31:
I had very high hopes that the ground would fix the issue - unfortunately, it made no difference.   :-(

jgyates, 2021-07-31:
Now that there is a ground have your tried swapping the Tx and Rx again?

eBoon123, 2021-07-31:
> Now that there is a ground have your tried swapping the Tx and Rx again?

Yes - tried both ways...

jgyates, 2021-07-31:
I am assuming your loopback was at the molex end of your cable, let me know if this is not the case.

Your logs show that you are getting 100% time out errors (i.e. nothing is returning from the controller). You could have a bad molex pin on your controller or a bad RS-232 driver or receiver on your controller. If you have a multimeter you could measure the voltage levels of the tx pin on the controller (the one that goes to the rx pin on the RS-232 connector). This should be a negative voltage relative to ground pin), generally between -3 and -12 volts depending on the type of transceiver. 

Also, I assume this is a new unit with the wifi module. is the wifi module unplugged? a picture of the connection may help.

eBoon123, 2021-07-31:
>I am assuming your loopback was at the molex end of your cable, let me know if this is not the case.

 Yes - that is the case

> Your logs show that you are getting 100% time out errors (i.e. nothing is returning from the controller). You could have a bad molex pin on your controller or a bad RS-232 driver or receiver on your [...]

 Will give this a go tomorrow

>Also, I assume this is a new unit with the wifi module. is the wifi module unplugged? a picture of the connection may help.

 I'll attach a picture tomorrow, but yes, this is a NEW unit with the WiFi module. The WiFi module is unplugged for right now - but I do intend to plug it back in once I get genmon working.

Thanks again for your help!

eBoon123, 2021-07-31:
> Yes - that is the case

> >> NVM - decided to do this tonight.. The WHITE wire is the ground coming from the controller (pin #2 on the molex connector) and the BLACK wire is the Tx coming from the controller (pin #8 on the molex connector).  This is showing POSITIVE 6.57 volts. 
[image]
Just for fun - here is molex pin #7 (Rx - red wire) and pin #2 (ground - white wire) showing -.4 volts
[image]

> > Also, I assume this is a new unit with the wifi module. is the wifi module unplugged? a picture of the connection may help.
> 
> >I'll attach a picture tomorrow, but yes, this is a NEW unit with the WiFi module. The WiFi module is unplugged for right now - but I do intend to plug it back in once I get genmon working.
>>>My connections:
[image]
[image]
Plugging here on the controller:
[image]

> 
> Thanks again for your help!

jgyates, 2021-07-31:
That voltage is correct (ground should go to the common on your meter so if you revers them you should get negative 6V. If you plug this in to your connector can you measure the -6V on the Rx pin? while you are at it you can check the voltage of the tx output on your RS-232. This should be in the same range (negative voltage when idle).

eBoon123, 2021-07-31:
> That voltage is correct (ground should go to the common on your meter so if you revers them you should get negative 6V.
  
  LOL - that's the problem with using a WHITE wire for the ground - didn't even notice I reversed it! :-)

>If you plug this in to your connector can you measure the -6V on the Rx pin? 
 
Like this? (sorry I'm a bit nooby with the multimeter):
[image]

>while you are at it you can check the voltage of the tx output on your RS-232. This should be in the same range (negative voltage when idle).

 Like this?
[image]

Thanks!

jgyates, 2021-07-31:
All the measurements should be on the green connector if you are measuring the RS-232 levels. If you measure on the pi side it will be 3.3v or zero volts only depending if it is a logic 0 or 1 respectively).

eBoon123, 2021-07-31:
> That voltage is correct (ground should go to the common on your meter so if you revers them you should get negative 6V. If you plug this in to your connector can you measure the -6V on the Rx pin? 

[image]

>while you are at it you can check the voltage of the tx output on your RS-232. This should be in the same range (negative voltage when idle).

[image]

Hope I got the measurements correct. Thank you again for your help and patience!

jgyates, 2021-07-31:
This looks good. If genmon is running do you see these values change (I am not sure that the fidelity of your multimeter will show this but it might). I would guess that you should see the values not be stable while genmon is running, possibly moving very briefly every 3 seconds (the length of the time out)or so. If you see the pi transmit fluctuate but not the receive then this puts the problem on the RS-232 side.

eBoon123, 2021-07-31:
OK - A couple of updates. When I plugged everything in this morning to take the above pictures - I left everything running while I went on an errand. Came back and realized that Genmon was actually working! (Yay!). So, of course, I had to shut it down and remove (only!) the molex connector from the generator so I could bring the box inside to measure for the hole I am going to cut to run wires. I then went out and put everything back and (of course!) I am now right back where I started - Not Receiving Data.

Weird - but I'm too surprised because I didn't change anything from yesterday when it was not working. So the mystery is why did it work 1 time and not the other 20 times I tried - sounds like something is flakey.

I did repeat the voltage measurements from above on the Rx and Tx pins - but this time my voltages are both about 3.99 - not sure if that is significant or not.

>If genmon is running do you see these values change (I am not sure that the fidelity of your multimeter will show this but it might).

Yes - but only very small changes on the third decimal point value.

skipfire, 2021-07-31:
I would think that it is some kind of cable issue.  It looks like you have a change in gauge from the molex wire to your extensions, that is the first spot I would check.  If you removed and reseated any molex pins you may want to recheck the seating on those.

I set mine up with the molex wires going directly into the serial screw terminals making a 4 inch cable, then used a serial extension cable for the distance, it avoided manual splices and had a better cable for the distance portion.

skipfire, 2021-07-31:
You might also check for any stray conductors at the screw terminals, maybe you have one that flared out instead of going into the screw terminal and is making contact with the other wire direction.

eBoon123, 2021-07-31:
> I would think that it is some kind of cable issue. 

Ding Ding Ding!
[image]

It's either these jumpers or the pins. Kept fiddling with these connections and finally, it's working! Of course the board could also be flakey, but I'll keep an eye on it and see if it keeps running.

Thank you @jgyates and @skipfire for your help and advice! Very excited to start playing with the app!!

skipfire, 2021-07-31:
I would suggest putting heatsinks on the 2 top chips since this will be outside in the heat and with the case pictures you have I'm guessing there is not much airflow.

eBoon123, 2021-07-31:
> I would suggest putting heatsinks on the 2 top chips since this will be outside in the heat and with the case pictures you have I'm guessing there is not much airflow.

Already ordered and should be here tomorrow. Hoping that with that box all buttoned up it doesn't get too hot in there!

skipfire, 2021-07-31:
It has been in the 90s here lately and I have my Pi in the generator and see CPU peaking around 65C, so it gets hot but not too hot.

jgyates, 2021-07-31:
I am glad it is working now. I am going to close this thread but feel free to comment here if you have related issues or questions.

closed 2021-07-31 by jgyates

jgyates, 2021-08-01:
@eBoon123 

I checked in a minor update that includes identification and fuel consumption of the Evolution 2.0 18kW model (V1.17.01).

eBoon123, 2021-08-01:
> @eBoon123
> 
> I checked in a minor update that includes identification and fuel consumption of the Evolution 2.0 18kW model (V1.17.01).

Awesome - just upgraded. I can now see values (0.00) instead of unknown.

Thanks again for this project - looking forward to playing with it more!

eBoon123, 2021-08-02:
I am thinking more and more that my TTL to RS232 board is bad - stopped receiving data again last night. I am currently using [this board](https://www.amazon.com/dp/B072KJSS5C/ref=sspa_dk_detail_3?psc=1&pd_rd_i=B072KJSS5C&pd_rd_wg=Nd6FS&pd_rd_r=VZ5RZZSGJ0JP5JMS1AYK&pd_rd_w=UqGO1). This is the board that was recommended as a second choice in the Hardware Wiki. Amazon currently does not have any stock for [the first board that was recommended](https://smile.amazon.com/gp/product/B00EJ9NAKA/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1).

Any suggestions on an alternate board? Would [this ](https://www.amazon.com/HiLetgo-MAX3232-Converter-Module-Serial/dp/B00LPK0Z9A/ref=asc_df_B00LPK0Z9A/?tag=hyprod-20&linkCode=df0&hvadid=390329205806&hvpos=&hvnetw=g&hvrand=12057979747322657643&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9010585&hvtargid=pla-906418052077&psc=1&tag=&ref=&adgrpid=79862087277&hvpone=&hvptwo=&hvadid=390329205806&hvpos=&hvnetw=g&hvrand=12057979747322657643&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9010585&hvtargid=pla-906418052077) work?

Thanks again for all the help!

jgyates, 2021-08-02:
Yes, that should work.

jgyates, 2021-08-02:
@eBoon123 

Since you have "Automated Feedback" enabled, your genmon installation sent a notice to me of an alarm log entry that the software has not seen before. If you can look at your log on your generator and let me know the exact text of the log with this date and time (08/01/21 21:24:39) I can add it to the software. I am guessing this is a new alarm code that is specific to Evolution 2, but that is a guess.

eBoon123, 2021-08-02:
I’m guessing this is it:

This is a snapshot from Mobile Link - is there somewhere else I should look
to get you the info?

Thanks

jgyates, 2021-08-02:
Images are stripped if you replay to the thread in email. You can post an image directly in the thread on github via this link in a browser:

      https://github.com/jgyates/genmon/issues/564

You can also go to your generator and select Sub Menus -> History -> Alarm Log then scroll thru until you see the entry with this date:

   08/01/21 21:24:39

Then just type in the text for that log entry into this thread.

eBoon123, 2021-08-02:
Looks like a 2 liner:
"WIRING ERROR"
"ERROR CODE 2098"

A little background - had a large storm here with a lot of lightning last
night. I saw a huge strike very near our house and noticed the status LED
on the generator was Red. Went out and reset and put the generator back
online - no further issues (so far).

I did a bit of research and found that one of the potential (non-wiring
error) ways to get this error, is a large nearby lightning strike - so
hoping that's was was the case.

Thanks again for all of your help and such an awesome project. Please let
me know if you have any questions or need me to look at anything further.

jgyates, 2021-08-02:
I checked in an update that should report this log entry correctly now. I did not update the version number so just go to the About page and update the software and your should be good.

eBoon123, 2021-08-02:
Looks good - thanks!