jgyates/genmon #566: Sourcing 3.3v power for the RS242<->module

evanevery, 2021-08-13:
This is more of a clarification/question rather than an "issue". Please let me know if there is a more appropriate place to post these types of things...

I'm working on getting all the power/data signals cleaned up to work with this project.  I may, in fact, just build a custom cable with an inline converter (in its own enclosure) to be able to feed multiple types of microcontrollers (Pi, Arduino, ESP32) in their own respective enclosures...

Anyway...  I understand the optimal signalling voltage (for the Pi) is 3.3v.  However, I see that the diagram/schematic posted for this device shows usage of a 5V power source for the RS232<->TTL converter.  I do see in the notes, mention of 5V vs 3.3V and even that 3.3v would be optimal (although 5V seems to be OK).

Why doesn't the diagram show the ideal source of power for the RS232<->TTL conversion module to be the 3.3V pin on the GPIO header? 

Am I missing something?

jgyates, 2021-08-13:
Hi @evanevery 

The reason why the note and the diagram do not match is that the diagram was provided to me by a user and it has not been updated since the note was created. I opted to keep the diagram instead of recreating it. If anyone wants to submit an updated diagram I would be happy to replace the existing one.

evanevery, 2021-08-13:
Here's an updated diagram.  Will this work?

[image]

jgyates, 2021-08-13:
Thanks! I updated the wiki with this diagram.

closed 2021-08-13 by jgyates