jgyates/genmon #567: Custom Genmon Hardware Available for Purchase

jgyates, 2021-08-20:
While you have the option of purchasing all of the components individually, there is an option for purchasing custom designed hardware that will simplify the hardware assembly process. More info is available [here](https://github.com/jgyates/genmon/wiki/2--Hardware#custom-hat).

toggledbits, 2021-10-30:
@skipfire I just purchased the ZGI-Gc HAT from you a couple of weeks ago. I note there's a button and what looks like a NeoPixel on the board. Is there a doc somewhere that would tell me where/how these can be used?

skipfire, 2021-10-30:
@toggledbits, the button and tri-color LED are for supporting an all in one GenMon unit I'm working on getting ready, and future revisions of the HATs may not have them which is why they are not documented anywhere.

Pin numbers (not GPIO numbers)
07 - momentary push button to ground
12 - BLUE (same as pin used by gengpioledblink extension)
19 - RED
23 - GREEN

bdwilson, 2022-01-27:
@skipfire do you have a 3D case designed for the Zero board?

bmg01907, 2022-08-15:
I purchased a GenPi fully assembled Generac generator monitor in 2019.  It uses an R-Pi 4 and works great.  It includes an illuminated pushbutton that when pressed resets the GenPi into AP mode to reconfigure the network settings.  The attached photo of the instructions describes this.

There are four wires connected to the pushbutton: two (white and red) that are connected to the R-Pi 4 GPIO pins 20 and 22, respectively.  Those wires appear to light up the pushbutton when there's a network connection.

The other two wires (brown and black) are connected to the normally open contacts on the pushbutton, so when the pushbutton is pressed they're shorted to provide a "reset" signal to the R-Pi.  Those two wires became disconnected from the R-Pi 4 GPIO header.

Does anyone know which GPIO pins should be connected to these brown and black wires?  I've reached out to the seller but haven't heard back yet.

[image]

jgyates, 2022-08-15:
Please use a new thread for new issues. Try GPIO3 (pin 5). This is likely using the add on [gengpio.py](https://github.com/jgyates/genmon/wiki/1----Software-Overview#gengpiopy-optional)

bmg01907, 2022-08-16:
Thanks.  I heard from the seller who said, "it should be GPIO7 (pin 26) and GND (pin 25)."  I've attached a photo for reference. 

[image]

basshook, 2023-04-14:
I wanted to give my endorsement to the PintSize.Me HAT. It absolutely makes the hardware setup of this project much simpler.
As far as support, @jgyates and @skipfire (PintSize.Me HAT) were great, helping to discover an unknown requirement setup in the 48kw, 4.5L controller. @skipfire sent me two versions of HAT to ensure there wasn't a hardware problem. Both HAT's worked flawlessly.

grsthegreat, 2023-09-08:
I have 5 to 6 people interested in buying a complete genmon ready to install unit, like i purchased years ago on ebay.  Anyone here doing this still? Thank you

skipfire, 2023-09-08:
Mine directly sold through pintsize.me website, the default payment is credit card through PayPal, but I do have some manual options as well.

grsthegreat, 2023-09-08:
I clicked the link, but dont see the complete genmon box similar to mine. What am i missing. The people ive spoken to have no idea how to set up rasp. Pi system….nor do i.Greg stevensonSent from my iPad

skipfire, 2023-09-08:
The preloaded unit would be what you want.  It includes a Pi 3A+, HAT, SD card with Pi OS and Genmon installed, power cable, and data cable.  I don't know what all was in the box you had, but configuring wifi is the only big part you would have to do.

jgyates, 2023-09-08:
I think what @liltux sold was a fully installed pi with an enclosure. He also had a handy web based wifi setup that would walk you through the process of connecting your wifi to the pi without having to log into the pi.

grsthegreat, 2023-09-09:
Ya, i tried emailing him at ***@***.*** as he stated, but have not heard back. Not sure if hes even interested in maxing them any longer. His product installed easy, especially for computer dummy like me.GregSent from my iPad

grsthegreat, 2023-09-09:
> The preloaded unit would be what you want. It includes a Pi 3A+, HAT, SD card with Pi OS and Genmon installed, power cable, and data cable. I don't know what all was in the box you had, but configuring wifi is the only big part you would have to do.

Yes, all i did was plug in included wire to accessory port, connect bat cables, and link to my wifi. Very easy. I suck with computers.

grsthegreat, 2023-09-09:
> The preloaded unit would be what you want. It includes a Pi 3A+, HAT, SD card with Pi OS and Genmon installed, power cable, and data cable. I don't know what all was in the box you had, but configuring wifi is the only big part you would have to do.

Is thebpreloaded unit still available. I dont see it in the pintsize.me website

skipfire, 2023-09-09:
Yes it is, https://pintsize.me/store/ols/products/generator-hats-and-phat/v/PGI-RPL is the link to it.  It is one of the form options.

grsthegreat, 2023-09-09:
Thanks, I'll check it out.

skipfire, 2023-09-15:
@grsthegreat I've got a new pre-built image that supports the WiFi configuration portal so you don't have to use the text file or GUI WiFi configuration.  If you have other absolute must-haves feel free to let me know and I will at least look into them.