jgyates/genmon #568: Startgenmon.sh executable.

lqallison, 2021-08-27:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

lqallison, 2021-08-27:
Installed from scratch today, found when trying to start genmon that 
/home/pi/genmon/startgenmon.sh was not executable. 
Fixed by chmon +x startgenmon.sh.

jgyates, 2021-08-28:
The bash script files in the repository are not executable when you check them out. The [instructions in the wiki](https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software) describe using this command first:

     bash ./genmonmaint.sh -i

The genmonmaint.sh script, if you use the "-i" option, for install, will update the permissions of both genmonmain.sh and startgenmon.sh to be executable.

jgyates, 2021-08-28:
I forgot to mention. If you use the "bash" command the script does not need to marked executable, so just run the above command first and you should be set.

closed 2021-08-28 by jgyates