jgyates/genmon #569: Genmon and Evo2 Controller Data Error: CRC check failed and loopback passes

gss2002, 2021-08-28:
Went through all of the doc just want to verify that this is a problem with either serial cable or controller as the loopback test passes.

### Actual Behavior

Unable to pass data from Evo2 to Genmon.py 

### Steps to Reproduce (including precondition)

Genmon with rPi to USB Serial to Controller. 
2021-08-28 15:05:47,619 : Error: timeout receiving slave packet for register 0236 Buffer: 7
2021-08-28 15:05:47,624 : Buffer = [0x9d,0x03,0xc2,0xf8,0xf8,0x6c,0x58]
2021-08-28 15:05:50,660 : Error: timeout receiving slave packet for register 0237 Buffer: 7
2021-08-28 15:05:50,666 : Buffer = [0x9d,0x03,0xc2,0xfc,0xfc,0x6c,0x58]
2021-08-28 15:05:53,700 : Error: timeout receiving slave packet for register 0239 Buffer: 7
2021-08-28 15:05:53,705 : Buffer = [0x9d,0x03,0xc2,0xf8,0x05,0xa8,0x5b]
2021-08-28 15:05:56,740 : Error: timeout receiving slave packet for register 023b Buffer: 7
2021-08-28 15:05:56,746 : Buffer = [0x9d,0x03,0xc2,0xf8,0xbe,0xe8,0x28]
2021-08-28 15:05:56,798 : Data Error: CRC check failed: 5701  f8fc
2021-08-28 15:05:56,804 : Error Receiving slave packet for register 023e
2021-08-28 15:05:57,865 : Data Error: CRC check failed: dc21  5868
2021-08-28 15:05:57,870 : Error Receiving slave packet for register 0209
2021-08-28 15:05:58,931 : Data Error: CRC check failed: dc21  5868
2021-08-28 15:05:58,936 : Error Receiving slave packet for register 020d
2021-08-28 15:06:02,971 : Error: timeout receiving slave packet for register 020f Buffer: 7
2021-08-28 15:06:02,976 : Buffer = [0x9d,0x03,0xc2,0xf0,0xc0,0x68,0x58]
2021-08-28 15:06:06,007 : Error: timeout receiving slave packet for register 0238 Buffer: 7
2021-08-28 15:06:06,012 : Buffer = [0x9d,0x03,0xc2,0xf8,0xf8,0x6c,0x58]
2021-08-28 15:06:09,049 : Error: timeout receiving slave packet for register 023a Buffer: 7
2021-08-28 15:06:09,056 : Buffer = [0x9d,0x03,0xc2,0xf8,0xf8,0x68,0x58]
2021-08-28 15:06:12,091 : Error: timeout receiving slave packet for register 023d Buffer: 7
2021-08-28 15:06:12,097 : Buffer = [0x9d,0x03,0xc2,0xe0,0xe0,0x68,0x58]
2021-08-28 15:06:15,134 : Error: timeout receiving slave packet for register 0241 Buffer: 7
2021-08-28 15:06:15,140 : Buffer = [0x9d,0x03,0xc2,0xf8,0xf8,0x6e,0x58]

### Your Environment

- Generator Model: G0071780
- Genmon Version: V1.17.01

jgyates, 2021-08-28:
Since you are getting CRC and timeout errors you likely have a wiring error on your cabling. [This thread](https://github.com/jgyates/genmon/issues/558) has a similar issue with pictures.

Let me know if this does not resolve your issue.

gss2002, 2021-08-28:
It was an issue on my side I had the serial ground connected to accessory power. I guess that's what happens when you are working on it till 1am :). I did try to use this with a Centos 7 VM and a Debian 10 VM on two difference VMware ESXi machines with a prolifics and a keytronics USB Serial Adapters just don't pass correctly and cause the loopback tests to fail at random. So I setup a rPi in my rack. I will send pics and provide doc if you want related to this setup.

Thanks again.. Also I assume these random ones are due to the known issue that is benign: #539 

2021-08-28 15:51:33,207 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-08-28 15:51:33,211 : Error Receiving slave packet for register 001e
2021-08-28 15:51:34,259 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-08-28 15:51:34,263 : Error Receiving slave packet for register 0001
2021-08-28 15:51:35,312 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-08-28 15:51:35,316 : Error Receiving slave packet for register 0053
2021-08-28 15:59:49,045 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-08-28 15:59:49,051 : Error Receiving slave packet for register 0039

closed 2021-08-28 by gss2002

jgyates, 2021-08-28:
yes, the modbus exceptions are expected with the Evo2.0 controller. I am glad it is working for you.