jgyates/genmon #570: Browser back button bypasses 2FA

djrallard, 2021-08-29:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

With 2FA enabled, the 2FA code should be required for login. A blank or invalid should deny login. 

### Actual Behavior

With 2FA enabled, 2FA can be bypassed with only user name and password required for login. 

### Steps to Reproduce (including precondition)

enter username and password at login screen and you are directed to the next screen to input your 2FA code. Without entering the code, press the back button in the browser and you will be logged in.

Tried from iPad, iPhone, Windows 10 laptop. 2FA code bypass via browser back button succeeded on all devices. 

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac 17kW
- Genmon Logs: Submit via the About page
- Genmon Version: 1.17.01

jgyates, 2021-08-29:
thanks for reporting this. I tested this and can duplicate the problem. I checked in a fix (V1.17.02) . This corrects the issue for me. Let me know how your testing goes.

djrallard, 2021-08-29:
The bug appears to be fixed with v1.17.02. Tested and now works as expected.

Thanks for an excellent project.

closed 2021-08-29 by jgyates