jgyates/genmon #571: Generac Extension Cable fitment (Not Molex mini-fit?)

evanevery, 2021-08-30:
I've got my Genmon installation up and running using a Pi4B and its great!  After some "connectivity" issues, I had to swap the TX and RX wires right at the Pi.  I'm not sure if its a doc/printing issue on the TTl/RS232 converter I'm using, or user error, but its working great now.

I also completely separated the RS232/TTL and Voltage conversion requirements from the Pi.  I built a separate "converter box" which takes the 4 wires from the generator (12v, Gnd, TX-RS232, RX-RS232) and converts them into a 6 wire output (12v, 5v, 3v3, Gnd, TX-TTL, RX-TTL).  I did this because eventually I would like to try and implement a similar project on an ESP32 based board with built in Cellular Connectivity.  I had to generate 3v3 (for the TTL/RS232 converter) and 5V (for the Pi) anyway so why not make all the voltages available on the output connector to maximize flexibility for future projects?

Here are the (two) voltage converters I am using (one set to 3v3 and the other to 5v):  [https://www.amazon.com/gp/product/B085T73CSD](url)

...and this is the TTL/RS232 converter: [https://www.amazon.com/gp/product/B091TN2ZPY](url)

ANYWAY!

I had the old Generac cellular mobile link system (EOL) still installed in my generator which already had a cable to merge the TX/RX and battery connections so I figured I would use it (instead of making my own cable).  (The Generac Part numbers for this harness is 0K2231 or 0K2220 depending on the length needed).  So this harness has WHAT LOOKS LIKE a 6-pin male Molex "mini-fit jr" connector on it.  (This connector normally plugs into the female connector on the back of the "mobile Link" cellular device).

So, I purchased a couple of 6-pin Molex Mini-Fit Jr connectors to make my cable.  (These are the same family of connector used to make the 8-pin variations used in this project to connect to the back of the Nexus/Evolution controllers).  Although the socket spacing on the connector seems to exactly match the mini-fit jr, the shape of the individual plastic "sockets" does not!  Some of those sockets have rounded corners and some are square to ensure the connectors are being mated properly...

I was able to manually "convince" the two connectors to mate, but they are not a proper match.  Anyone know what type of connector that might be on the end of those harnesses (so I can make a perfect cable)?  They are close to "mini-fit jr" but not an exact match...

skipfire, 2021-08-30:
Any reason to not use an LTE had for the Pi?  Could do it with a LTE Feather board too.

evanevery, 2021-08-30:
I actually prefer to write directly to the microcontroller in C (and skip the O/S altogether).  I already have designed several variations of ESP32/Arduino boards with onboard cellular chips (SIMCOM 800/900/5320/7200/7600).  Arduino/ESP32 is actually my personal design preference and I have many Open Source libraries already developed to support these chips in the GSM environment.

Anyone interested can find a description of all my boards (and open source software) at the following link:  [https://www.haute-solutions.com/](url) ...  ;-)

jgyates, 2021-08-31:
Unfortunately I do not. The original cable I made used the part numbers for mini fit listed in the wiki however I now use a few of [these](https://www.amazon.com/gp/product/B000M802RG) connected together to make a cable.

evanevery, 2021-09-01:
So, I think I found the connector type for the end of the "Mobile Link" harness (for anyone who is replacing their Mobile Link unit).

It appears to be a 6-pin TE Connectivity "Mini-Universal MATE-N-LOK Connector".  The individual socket shapes and the pin spacing all look to be correct.  These should be the Mouser and TE Connect part numbers for the connector/pins to mate onto the end of the Mobile Link Harness (where the Mobile Link unit would normally attach).  I have a couple sets of these ordered and will confirm once I receive them:

Pin & Socket Connectors CAP HOUSING 6P (x1)
Mfr. #:                 172331-1                                        
Mouser #:            571-1723311

Pin & Socket Connectors PIN CONTACT 22-18AWG Gold (x6)
Mfr. #:                  1-770987-0                                         
Mouser #:             571-1-770987-0

closed 2021-09-02 by jgyates