jgyates/genmon #572: RPi USB and GenMon

swvalenti194, 2021-09-09:
Not sure if it's an issue but rather ask...I am trying to run genmon on a rpi3b+ from a usb drive but os never loads.  I was able to do this on other rpi I have so just wondering if genmon prevents that?  Anyone use usb and genmon?

liltux, 2021-09-09:
So you are running Rasbian or some linux arm distribution from USB drive and not the MicroSD?
What is not loading the OS or linux distro or Genmon is not loading?

swvalenti194, 2021-09-09:
Yes running buster from usb not micro.  The OS isn't loading which is odd cause I have other rpi that load fine.  Didn't know if a peripheral was not allowing it too like the serial to ttl converter.

liltux, 2021-09-09:
I cannot imagine any of the serial settings affecting the boot loading of the linux kernel.  If anything the linux kernel should still boot.  Did you take this USB drive from one Pi to a different one and now it does not load?  Or did you burn an image to USB with Genmon already installed?  More than likely and issue with the imaging of the USB drive.

swvalenti194, 2021-09-09:
Good to know, I used SD card copier to USB on several units with no issue but genmon one seems to fail.  Oh well thanks for checking in

jgyates, 2021-09-09:
Genmom has nothing that would keep the system from boot from usb. You do have to make the boot config file specify usb. This article details how to do it.

https://thepi.io/how-to-boot-your-raspberry-pi-from-a-usb-mass-storage-device/

closed 2021-09-09 by swvalenti194