jgyates/genmon #573: "System Initializing. Not receiving data" on 2008 controller

jcleek, 2021-09-09:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

Without loopback
> Loopback testing for serial port /dev/serial0...
> Serial port opened
> write data: sent test string
> waiting to received data....
> FAILED: Sent data does not match receive. Received 0 bytes`

With loopback
> Loopback testing for serial port /dev/serial0...
> Serial port opened
> write data: sent test string
> waiting to received data....
> PASSED! Loopback successful

I am actually able to start the generator from the web interface but could not shut it off. Time restriction maybe?

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Initialization and receiving data. I have a pre-2008 model as indicated here:

> Some 2008 and Earlier Residential Models- Some models from around 2008 have a controller that looks a lot like a Nexus controller, except they do not have the word "Nexus" on the front. This picture shows the controller. Testing has shown that these controllers respond over the serial link in a non-modbus compatible way. To support these controllers you must set "Modbus slave transmit address" on the advanced setting page in the web interface (or response_address in /etc/genmon.conf).

I have been through every screen and the genmon.conf file and am unable to find these settings anywhere. I do not see an ADVANCED settings page or anywhere to enable such a page, nor do I know what address it should be set to. I did find a thread where someone was told 41 was the right setting and I would try that if I knew where to put it.

### Actual Behavior

Red bar at bottom indicating "System Initializing. Not receiving data".

### Steps to Reproduce (including precondition)

Power up the Pi

### Screenshot or Pictures relating to the problem (if possible)

[image]

[image]

### Your Environment

- Generator Model: 2008 17Kw 0055040
- Genmon Logs: Submit via the About page
- Genmon Version: V1.17.02
- I have the generator HAT and cable from https://pintsize.me/store/ols/products/generator-hat/v/PGI-Gc
- Raspberry Pi 3B+

liltux, 2021-09-09:
@jcleek The advanced page is accessed via the top left corner:
[image]
The wrench and screw driver.  One click opens the registers page, 2 clicks opens the advanced page.  I have noticed if Comms is not working some of the pages are laggy so be careful not to over/under click.

liltux, 2021-09-09:
From #316 
[image]

jcleek, 2021-09-09:
Ok, didn't realize when you clicked the wrench/screwdriver icon it changed and the advanced page.

Put in 41 for "Modbus slave transmit address" and appears to be working. Thank you!

liltux, 2021-09-09:
Outstanding!

closed 2021-09-09 by jcleek

jgyates, 2024-10-04:
@jcleek 

do you see anything like the behavior described in this thread: https://github.com/jgyates/genmon/issues/1191

jcleek, 2024-10-04:
I responded in that thread. If you want logs or anything let me know.