jgyates/genmon #574: 3D printed case for Pi4B and Genmon Hat

evanevery, 2021-09-13:
FYI - If anyone is interested:

Here is a download link for a 3D printed case I designed to house a Pi4B with the PintSize.Me HAT attached:

[https://www.thingiverse.com/thing:4964357](url)

jgyates, 2021-09-13:
Very nice! I will add this to the wiki

jgyates, 2021-09-13:
The hyperlink is not showing up correctly. https://www.thingiverse.com/thing:4964357

jgyates, 2021-09-13:
https://github.com/jgyates/genmon/wiki/2--Hardware#custom-3d-printed-enclosure

skipfire, 2021-09-14:
@evanevery very cool!  Can't wait to get home and try printing one on Saturday.

evanevery, 2021-09-14:
The trick is having a good collection of small screws!

dmjpixel, 2022-07-17:
Just received latest Pi Hat - printed 3D case but looks to be for the original hat and not the updated version (opening is not aligned for the barrel plug for example).  Are there plans to update the 3D files by chance; I am not good at designing.  Thanks - very excited to try this project with my new gen.

skipfire, 2022-07-17:
At this point I would guess there is not a plan to do it.  @evanevery did the design on his own after purchasing.  I'm also not great at doing case design.  The new board also would have some functional differences as it would need to allow for the fan and airflow.

davisgoodman, 2022-07-17:
I wouldn't mind giving it a shot but not having the Pi Hat physically is kind of hard...

nickabocker171, 2023-08-30:
I don't any place to upload these but wanted to share. I had a good family member help me out with making some changes to the original drawings provided here. I hope @evanevery is alright with this but here are updated files for the pi4B, as well as files for a pi3b and a pi3a. [https://drive.google.com/drive/folders/11xLq5BuweIsJon-HDvMjko1bGZA3YWuY?usp=sharing](pi stl files) These are all designed and work with the newer hats.

[image]

closed 2024-01-09 by jgyates