jgyates/genmon #575: Current Transformers Amazon

philruel, 2021-09-18:
Hi,
This post is mainly to have more information from sefs85, grsthegreat and others who bought Current Transformers from Amazon because the 22 kW air cooled Evolution 1 are not equipped.  The links both of you gave were for model # PZCT-2.
 
I also bought 2x PZCT-2 from Amazon.  I wanted to test them before installing in my 22 kW so I hooked them to my clothes dryer, specs 230 V and 24 Amps.
Results with my clamp meter, average of 23.6 Amps. All is good here.
I hooked both my PZCT-2 to the wires of my dryer and set my multimeter to VAC and got an average of 15.6 VAC
 
In the Generac Diagnostic Repair Manual p.80, 25 Amps = 0.380 VAC.  I'm way over this value.
 
What is wrong or what am I not doing correctly? I don't want to fry my Generac controller.
 
Also, can either wire from CT be connected to either terminal in the controller?
 
Thank you

•	Generator Model: {G0065520} 22kW Evolution 1
•	Genmon Version: {1.17.02}

grsthegreat, 2021-09-23:
i dont know the answer. i never tested the ct's/ i just installed them and they worked 100%

philruel, 2021-09-24:
Thank you.

Anyone else?

jgyates, 2021-10-06:
I am going to close this issue due to a lack of response but feel free to post here if there are any updates.

closed 2021-10-06 by jgyates