jgyates/genmon #576: Notification Emails without any detail reason.

sambeetm, 2021-09-23:
Hi, My genmon is up and running for almost 2 weeks. Occasionally, I am getting following 2 emails. Looking into the same, it says some Alarm. Then it says "The system is not in alarm." What exactly is going on here ? No further details. How do I get the details ?

Email 1
--------
Time: 12:26:57 PM
Date: Thu 23-Sep-2021

NOTE: This message is a notice that the state of the generator has changed. The system is not in alarm.

Status :

    Engine :
        Switch State : System in Alarm
        Engine State : Off - Ready
        Battery Voltage : 12.90 V
        RPM : 0
        Frequency : 0.00 Hz
        Output Voltage : 0 V
        Output Current : 0.00 A
        Output Power (Single Phase) : 0.00 kW
        Active Rotor Poles (Calculated) : 0

    Line :
        Utility Voltage : 243 V
        Utility Max Voltage : 8224 V
        Utility Min Voltage : 234 V
        Utility Threshold Voltage : 156 V

    Last Log Entries :

        Logs :
            Alarm Log : 08/02/21 11:46:54 Low Battery : Alarm Code: 0000
            Service Log :
            Run Log : 09/16/21 11:54:48 Stopped - Auto

    Time :
        Monitor Time : Thursday September 23, 2021 12:26:56
        Generator Time : Thursday September 23, 2021 12:26

No Alarms: 0001:00000074

Email 2
---------
Time: 12:27:09 PM
Date: Thu 23-Sep-2021

NOTE: This message is a notice that the state of the generator has changed. The system is not in alarm.

Status :

    Engine :
        Switch State : Auto
        Engine State : Off - Ready
        Battery Voltage : 12.90 V
        RPM : 0
        Frequency : 0.00 Hz
        Output Voltage : 0 V
        Output Current : 0.00 A
        Output Power (Single Phase) : 0.00 kW
        Active Rotor Poles (Calculated) : 0

    Line :
        Utility Voltage : 243 V
        Utility Max Voltage : 8224 V
        Utility Min Voltage : 234 V
        Utility Threshold Voltage : 156 V

    Last Log Entries :

        Logs :
            Alarm Log : 08/02/21 11:46:54 Low Battery : Alarm Code: 0000
            Service Log :
            Run Log : 09/16/21 11:54:48 Stopped - Auto

    Time :
        Monitor Time : Thursday September 23, 2021 12:27:07
        Generator Time : Thursday September 23, 2021 12:27

No Alarms: 0001:00000000

jgyates, 2021-09-23:
@sambeetm 

The software will notify via email when the engine / status register changes state. These two emails that you are seeing is a result of that. The engine / status register modbus register is 0001. The first email is this register changing to 0x74 and the second is the register returning to the normal / ready state (0x00). This is the first time I have seen this value (0x74) in the status register. I am assuming you have an Evolution 2.0. I have seen a 0x78 status (no wifi module). I am guessing that this is related to with wifi connectivity in some way (it could be failing to check for firmware updates or something like that). The controller would probably put an error message on the LCD screen but I am guessing it is for a short period of time so you should have to catch it doing it in the act. If you want to submit your logs on the About page (assuming you have outbound email enabled) I can see if there are any additional hints that may help.

What is the wifi signal strength on your controller?

Also, what version of the Evolution 2.0 firmware do you have 1.11 or 1.12? This could be new behavior for 1.12.

One thing that "might" work is the "Ignore Unknown Values" setting on the advanced page. Access the advanced page by double clicking the gear icon on the main web interface. Enable this and save the settings. Sometimes the evolution 2.0 controller would output garbage values in register 0001 and this setting ignores that value, however my guess is that this will not fix the issue as that setting was targeting unused bit fields and the value of 0x74 is using bit fields that are normally used in status / alarm values.

sambeetm, 2021-09-23:
I just sent the logfile using the "Submit Log" button. Let me know if you need any other file.
Just to mention that, I have both Generac WifI and Genmon interface setup working. I will get the wifi Signal level and firmware version when I go out. I assume its 1.12 as it was recently delivered(on Aug) and installed.

jgyates, 2021-09-23:
I got your logs, looks like you have V1.11. Nothing out of the ordinary. Based on the registers you submitted it looks like your signal strength on the controller wifi is -66dBm which is marginal. Improving this may help. If you are getting a lot of these messages I can add an option to ignore these conditions. Normally the software sends a message for any change in  register 0001.

sambeetm, 2021-09-23:
Ok. Thanks for the explanation. I have received these emails 3 times in last 2 weeks. I will keep monitoring for few more months and update you.

sambeetm, 2021-09-26:
Today I got another code. After 11 sec, it was cleared. Its 2020. What is this for ?

Time: 01:59:24 PM
Date: Sun 26-Sep-2021

NOTE: This message is a notice that the state of the generator has changed. The system is not in alarm.

Status :

    Engine :
        Switch State : System in Alarm
        Engine State : Off - Ready
        Battery Voltage : 13.70 V
        RPM : 0
        Frequency : 0.00 Hz
        Output Voltage : 0 V
        Output Current : 0.00 A
        Output Power (Single Phase) : 0.00 kW
        Active Rotor Poles (Calculated) : 0

    Line :
        Utility Voltage : 243 V
        Utility Max Voltage : 8224 V
        Utility Min Voltage : 234 V
        Utility Threshold Voltage : 156 V

    Last Log Entries :

        Logs :
            Alarm Log : 08/02/21 11:46:54 Low Battery : Alarm Code: 0000
            Service Log :
            Run Log : 09/16/21 11:54:48 Stopped - Auto

    Time :
        Monitor Time : Sunday September 26, 2021 13:59:21
        Generator Time : Sunday September 26, 2021 13:59

No Alarms: 0001:00002020

jgyates, 2021-09-26:
That values is consistent with number 11 under the Known Issues wiki page:

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

common values that are seen in this condition are 00002020, 20200000, 00003f3d and 3f3d0000.

jgyates, 2021-09-26:
To give a bit more data, alarm and status in register 0001 are typically in this format:

00xx00yy

whey xx is the switch state and yy is the engine state or current alarm. The values do not fit this pattern, however the 00000074 does which makes me thing that the 74 is an actual alrm but the other values are and artifact of the Evo 2.0 firmware since these values do not occur on other controllers that share the same alarm codes.

jgyates, 2021-10-14:
I am going to close this issue for now. If you have any additional data let me know via this thread.

closed 2021-10-14 by jgyates